StephanJoubert/home_assistant_solarman #504: Way to control Deye Inverter

Ramaddan, 2024-01-30:
Hi,

First, one of the best Home Assistant addons out there for me.

I am using a Deye inverter SUN-8K-SG01LP1

I was wondering if there is a way to control anything on the inverter, as in change any setting from Home Assistant.

If there is, then a person can do many interesting things though Home Assistant.

Thanks

ChrisBGL, 2024-02-01:
Yes, please integrate control. F.e. (de)active time of use in inverter config.

Ramaddan, 2024-02-04:
Yes, one idea in mind is to be able to turn ON and OFF the gen force, remotely

Currently, have to go down and up a building in case I need to force turn ON the generator

krasnoukhov, 2024-06-05:
I came up with a config to turn gen on/off for the deye invertor. I chose `custom_parameters.yaml` for the configuration, and changed the file slightly to this:

```yaml
requests:
  - start: 0x0003
    end: 0x0070
    mb_functioncode: 0x03
  - start: 0x0096
    end: 0x00f9
    mb_functioncode: 0x03
  - start: 0x00FA
    end: 0x0146
    # end: 0x0117
    mb_functioncode: 0x03

parameters:
 # ... all the parameters from deye_hybrid ...
 - group: Settings
   items:
    - name: "Grid Charge Ampere"
      class: "current"
      uom: "A"
      scale: 1
      rule: 1
      registers: [0x00E6]
      icon: 'mdi:current-dc'
    - name: "Gen Force"
      class: ""
      uom: ""
      state_class: ""
      scale: 1
      rule: 1
      registers: [0x0146]
      icon: 'mdi:generator-stationary'
      lookup:
        - key: 0
          value: "Off"
        - key: 8192
          value: "On"
```

`requests` needs a change to be able to query 0x0146 which is "gen force" option. And then I have template switch that uses the service to turn on/off:

```yaml
switch:
  - platform: template
    switches:
      gen_force:
        friendly_name: "Deye Gen Force"
        value_template: |
          {{ is_state("sensor.deye_gen_force", "On") }}
        turn_on:
          service: solarman.write_multiple_holding_registers
          data:
            register: 326
            values: "{{ [8192] }}"
        turn_off:
          service: solarman.write_multiple_holding_registers
          data:
            register: 326
            values: "{{ [0] }}"
        icon_template: >-
          {% if is_state("sensor.deye_gen_force", "On") %}
            mdi:engine
          {% else %}
            mdi:engine-off
          {% endif %}
```

Ramaddan, 2024-06-07:
Sorry, do I make the file custom_parameters.yaml, or is it supposed to be there?

If I make it, how do I make it recognizable to the system? Where do I call it?

Thanks

krasnoukhov, 2024-06-08:
[image]

You can use "configure" for your existing integration to switch to custom parameters config. This is a file you edit manually, in solarman custom component directory

Ramaddan, 2024-06-09:
Ok. Will try it when I am free. Thanks

Ramaddan, 2024-09-06:
Hi. I was very busy for a long time from being able to try anything.

I apologize for that.

I am finally going to try right now.

I made my own custom parameter config

I just need to know if I also need to add the template swith in the same file, ex: custom_parameters.yaml

Or does it go into the home assistant configuration.yaml file?

Is it possible to add it to the same file without having to go to configuration.yaml?

Thank you very much for your help

krasnoukhov, 2024-09-09:
> Or does it go into the home assistant configuration.yaml file?

Yes, exactly

> Is it possible to add it to the same file without having to go to configuration.yaml?

No, the only way to set this up is via configuration.yaml. It's not currently possible via UI either

krasnoukhov, 2024-09-09:
Weird, I can't find any reference to `wait_for_attempts` anywhere. You can try running `solarman.write_multiple_holding_registers` from the developer tools -> actions tab, see if it's working for you?

Ramaddan, 2024-09-12:
I tried what you suggested, and it was a bit finicky but worked

So it worked from the developers page, but not the button

I will try to look more into it when I have time

If I understood the code more it would have been easier

Out of curiosity, and to better understand, how did you get the turn ON and OFF value?

I am not using the original Stephan code, but a fork that is currently being maintained by davidrapan, if that makes any difference

Thanks

Ramaddan, 2024-09-12:
I kind of understood your code, but I just did not understand how you got the following numbers

0x0146 and 8192

krasnoukhov, 2024-09-12:
I used web version of solarman app, reverse engineered it by changing things and then looking into the requests it made

PV-Joe, 2024-09-12:
Did you guys try this Solarman integration yet?

It serves more sensors and configurations. It's also maintained regularly and has excellent support.

https://github.com/davidrapan/ha-solarman

Ramaddan, 2024-09-13:
> I used web version of solarman app, reverse engineered it by changing things and then looking into the requests it made

Ok. Got it 👍

Ramaddan, 2024-09-13:
> Did you guys try this Solarman integration yet?
> 
> It serves more sensors and configurations. It's also maintained regularly and has excellent support.
> 
> https://github.com/davidrapan/ha-solarman

Yes, I mentioned it here as well

krasnoukhov, 2024-09-13:
@PV-Joe Haha not sure if it's nice of you to slide into the comments promoting the competing integration project 😅 

But anyhow I dug into davidrapan's implementation and it seems to be more robust in terms of supporting various controls

Just switched my setup to that one personally, as well as submitted the PR to enable generator control and Ampere charge configuration: https://github.com/davidrapan/ha-solarman/pull/154

PV-Joe, 2024-09-13:
> @PV-Joe Haha not sure if it's nice of you to slide into the comments promoting the competing integration project 😅 

I think there is no competition in open source projects. Anyone can copy, fork and modify and there is no loss whatsoever, right?

krasnoukhov, 2024-09-13:
Yeah I know, just kidding