StephanJoubert/home_assistant_solarman #507: SN 39XXXXXXX (protocol v4)

romainlouvet, 2024-02-06:
Hello, 

I try to init this integration for my deye controller SN : 39XXXXXXXX but it doesn't work  :/

in this doc -> 

> Check the version of the solarman logger. If the serial number starts with 17xxxxxxx, 21xxxxxxx or 40xxxxxxx (protocol V5), the component should work. If not, you may need to try the component for V4 of the protocol mentioned above.

Where can I find this v4 component?

thanks

real-intruder, 2024-02-18:
Same Problem
[image]
[image]

real-intruder, 2024-02-18:
https://github.com/StephanJoubert/home_assistant_solarman/issues/367

It seems there is a solution with a different.yaml configuration for the Solarman Addon instead of  2mptt yaml

https://github.com/StephanJoubert/home_assistant_solarman/issues/367#issuecomment-1708839274

meerlaender, 2024-02-26:
The same Problem  SN 39XXXXX

what is the V4 must i use an older Solarmann Version or wait for a newer? 
[image]

THX for answer

Jaapk, 2024-02-28:
I seem to run into the same issue - any solution?

wandi34, 2024-03-14:
Same problem here

robertugo2, 2024-05-17:
I think author of this repo is suggesting to use 1.4.x version of this custom component. I noticed, that 1.4.x is working fine white 1.5.x is not working at all. I have Sofar hybrid inverter.

slyxy21, 2024-11-11:
How can I change protocol v5 to v4 ?

Addy0775, 2024-11-12:
Hi, my deye has a 23x number but its still disconnected. Tried with Version 1.5 and 1.4.3 with sg04lp3.yaml config. Ist there any other way?
Thanks