StephanJoubert/home_assistant_solarman #515: 2024.2 Not working

sdhein, 2024-02-12:
Solarman not working after 2024.2 core upgrade

rene-mt, 2024-02-13:
Same here. No connection anymore to Deye inverter after upgrade to HA 2024.2

81ondo, 2024-02-19:
How to fix it?

brownstone666, 2024-02-19:
Same here. Deye String model:

```
Logger: homeassistant.config_entries
Source: config_entries.py:605
First occurred: 10:29:47 (1 occurrences)
Last logged: 10:29:47

Error unloading entry 192.168.3.250 for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 605, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 28, in async_unload_entry
    hass.data[DOMAIN].pop(entry.entry_id)
    ~~~~~~~~~^^^^^^^^
KeyError: 'solarman'
```

PV-Joe, 2024-02-19:
Everything’s fine here with Version 1.4.3.

route66por, 2024-02-26:
I also have version 1.4.3 and no problems. And as i can see, i will still be on it for longer.

brownstone666, 2024-02-27:
> Same here. Deye String model:
>
> ```
> Logger: homeassistant.config_entries
> Source: config_entries.py:605
> First occurred: 10:29:47 (1 occurrences)
> Last logged: 10:29:47
>
> Error unloading entry [...]

I downgrade to version 1.4.3 and working like a charm