StephanJoubert/home_assistant_solarman #519: I have Sofar  5KTLM-G3-A with logger serial number 27

sateeshruttala, 2024-02-17:
I have used sofar_lsw3.yaml and wifi kit,still it says disconnected,can you please help me here

81ondo, 2024-02-18:
Same here.
Could you include your home-assistant.log?

Picard2002, 2024-02-25:
I have managed to obtain some data from this same investor in the following way.
with file editor create a new yaml file in custom_components/solarmar/inverter_definitions folder and save it as custom_parameters.yaml .
Next, open the definition for azurro and copy all the content, paste it in the file you created, when you configure the addon, fill in all the data and at the end put custom_parameters.yaml, restart home assistant and you should start getting data

kiwibhoy, 2024-06-08:
Where do you find definition file for azurro?