StephanJoubert/home_assistant_solarman #522: all entities unknown

Lyckell, 2024-02-22:
Have tried to get this integration to work with a SUN-10K-SG04LP3-EU but all entities in shows "unknown".

Does anyone know of a solution for this?

sebastiansiede, 2024-02-23:
I have the same problem with DEYE-SUN-M60G3-EU

cyberdreamandroid, 2024-02-23:
Same for me. Logger serial starts with 4, according to doc, this should be supported right?

Lyckell, 2024-02-23:
The serial for my logger starts with 27

oetiker, 2024-02-24:
I think the problem is that the extension broke with the 2024.02 release of homeassistant

eddyheide, 2024-02-24:
Same for me all entities in shows "unknown".
I have 3 phase Sofar

Karsto-sys, 2024-02-25:
I have the Same Problem

miralpeix, 2024-02-27:
My serial starts with 23****. All in "unknown". How can use the mentioned V4?

mmmmm009, 2024-02-27:
Same here. 
Newest HA.

IP:192.168.0.158
SN:17......
port:8899
1
deye_hybrid.yaml

route66por, 2024-02-28:
My suggestion is to downgrade do version 1.4.3. 
I had the same issues and now its all ok. I will stay on that version for longer.

cyberdreamandroid, 2024-02-28:
> My suggestion is to downgrade do version 1.4.3. I had the same issues and now its all ok. I will stay on that version for longer.

This is sad, isn't this developed any longer? How to downgrade?

route66por, 2024-02-28:
> > My suggestion is to downgrade do version 1.4.3. I had the same issues and now its all ok. I will stay on that version for longer.
> 
> This is sad, isn't this developed any longer? How to downgrade?

https://hacs.xyz/docs/faq/select_version/

route66por, 2024-02-28:
> > > > My suggestion is to downgrade do version 1.4.3. I had the same issues and now its all ok. I will stay on that version for longer.
> > > 
> > > 
> > > This is sad, isn't this developed any longer? How to downgrade?
> > 
> > 
> > https://hacs.xyz/docs/faq/select_version/
> 
> Thanks, but seems not to work, whenever I select 1.43 in select box, it refreshes and shows 1.5.1 again.

Try to remove the integration first, restart HA and then  install it again but this time install the 1.4.3 version.

djjassu, 2024-02-28:
Downgrade to 1.4.3 did not help here. Still all unknown...

cyberdreamandroid, 2024-02-28:
> Downgrade to 1.4.3 did not help here. Still all unknown...

Downgrade worked for me after a restart.

djjassu, 2024-02-29:
> > Downgrade to 1.4.3 did not help here. Still all unknown...
> 
> Downgrade worked for me after a restart.

Ok, did you also downgrade the HA?

My version:

Core
2024.2.5
Supervisor
2024.02.0
Operating System
12.0
Frontend
20240207.1

cyberdreamandroid, 2024-02-29:
Downgraded version 1.4.3 with current HA 2024.2.5 works.

cyberdreamandroid, 2024-02-29:
After another reboot, again "Unknown" even with 1.4.3. As the developer doesn't seem to be interested to fix this, I will develop my own integration for this.

eddyheide, 2024-02-29:
Please keep us posted if you get you own integration working!

astro0302, 2024-02-29:
Downgraded version 1.4.4 with current HA 2024.2.5 also works for me!

egennric, 2024-03-03:
1.5.1

Core 2024.2.5
Supervisor 2024.02.1
Operating System 12.0
Frontend 20240207.1

SUN-M80G3-EU-Q0
S/N 23..
works with both
deye_4mppt.yaml or deye_2mppt.yaml

Problem starts after upgrade firmware from *1.57*  to  MW3_16U_5406_2.32-D1

after some hours I run in this "unknown" problem
but resolve ist by remove the device and add it again with IP + S/N
the it runs for some hours again 
it stops working while the inverter is still online and reporting to https://globalhome.solarmanpv.com/
once "unknown" nothing but re add the device can fix it

miralpeix, 2024-03-04:
I have no way to work. I have tried all versions of the integration... my SN is 23**** and there is no way. any ideas?

superquale, 2024-03-18:
Logger: custom_components.solarman.solarman
Quelle: custom_components/solarman/solarman.py:88
Integration: Solarman (Dokumentation, Probleme)
Erstmals aufgetreten: 09:35:39 (92 Vorkommnisse)
Zuletzt protokolliert: 09:58:10

Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid start or end values]
Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid sequence number]

*****
In all Versions of SOLARMAN, i get this error

schlaufon, 2024-03-19:
> I have the same problem with DEYE-SUN-M60G3-EU

@sebastiansiede  see issue #467