StephanJoubert/home_assistant_solarman #524: Support for Pylontech Force H2 battery

pavelmaca, 2024-02-23:
Hi,
first of all, thans for great HA integration.

Pylontech Force H2 battery pack uses Solarman v5 logger, so i checked available addresses on local tcp/modbus using [jmccrohan/pysolarmanv5 register_scan.py](https://github.com/jmccrohan/pysolarmanv5/blob/main/examples/register_scan.py). 

Found valid addresses in range of 03972-06911 (scanned 0-9999)

Not sure if, everything is correct. Must do some more testing and observation, since documentation for this is none.

This is my `custom_parameters.yaml` for standalone battery:
```yaml
requests:
  # Brand and Device name
  - start: 4096
    end:  4105
    mb_functioncode: 0x03

  # Device SN
  - start: 5200
    end:  5207
    mb_functioncode: 0x03

  # Cycle count
  - start: 5128
    end:  5128
    mb_functioncode: 0x04

  # Pack, Module, Cell count
  - start: 5174
    end:  5176
    mb_functioncode: 0x04

  # Pack 0-3 Voltage
  - start: 5216 
    end:  5219
    mb_functioncode: 0x04
    
  # Pack 0-3 Temperature
  - start: 5296 
    end:  5299
    mb_functioncode: 0x04
 

parameters:
 - group: Basic information
   items:
    - name: "Device SN"
      class: ""
      state_class: ""
      uom: ""
      scale: 1
      rule: 5
      registers: [5200,5201,5202,5203,5204,5205,5206,5207]
      isstr: true

    - name: "Brand"
      class: ""
      state_class: ""
      uom: ""
      scale: 1
      rule: 5
      registers: [4096,4097,4098]
      isstr: true

    - name: "Device Name"
      class: ""
      state_class: ""
      uom: ""
      scale: 1
      rule: 5
      registers: [4101,4102,4103,4104,4105]
      isstr: true

    - name: "Battery Module (series)"
      class: ""
      state_class: ""
      uom: ""
      scale: 1
      rule: 1
      registers: [5174]

    - name: "Battery Cell (series)"
      class: ""
      state_class: ""
      uom: ""
      scale: 1
      rule: 1
      registers: [5175]

    #- name: "Battery Pack (parallel)"
    #  class: ""
    #  state_class: ""
    #  uom: ""
    #  scale: 1
    #  rule: 1
    #  registers: [5176]

 - group: Battery pack
   items:
    - name: "Cycle Times"
      class: ""
      state_class: ""
      uom: ""
      scale: 1
      rule: 1
      registers: [5128]
      icon: 'mdi:battery-heart'

 - group: Battery Module 0
   items:
    - name: "Battery Module 0 Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.01
      rule: 1
      registers: [5216]
      icon: 'mdi:battery'
      
    - name: "Battery Module 0 Temperature"
      class: "temperature"
      state_class: "measurement"
      uom: "°C"
      scale: 0.1
      rule: 2
      registers: [5296]
      icon: 'mdi:thermometer'
      
 - group: Battery Module 1
   items:
    - name: "Battery Module 1 Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.01
      rule: 1
      registers: [5217]
      icon: 'mdi:battery'
      
    - name: "Battery Module 1 Temperature"
      class: "temperature"
      state_class: "measurement"
      uom: "°C"
      scale: 0.1
      rule: 2
      registers: [5297]
      icon: 'mdi:thermometer'
      
 - group: Battery Module 2
   items:
    - name: "Battery Module 2 Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.01
      rule: 1
      registers: [5218]
      icon: 'mdi:battery'
      
    - name: "Battery Module 2 Temperature"
      class: "temperature"
      state_class: "measurement"
      uom: "°C"
      scale: 0.1
      rule: 2
      registers: [5298]
      icon: 'mdi:thermometer'
```

**Some another interesting addresess:**

```
Individual cell temperature? (got same values for all cells in module)
6144-6173 (module 0)
6174-6203 (module 1)
6204-6233 (module 2)

Individual cell voltage?
5376-5465 (3*30 by module)

Device datetime?
4320-4325
```

robin73, 2024-04-06:
Hi, 

nice work. Your custom_parameters.yaml working with my pylontech force H1 !! (Firmware versionMW3_15_B505_1.27
). Logger SN: 40xxxxxxxx

Thanks

gec75, 2024-07-24:
It also works for me... (Force H2 V2).
I modified for 5 Modules and changed Charge/Discharge (Today/Total) registers to 32 bit (rule: 3)

arizzi, 2025-02-08:
I'd like to monitor the Force L2 that I think share the same logger, but the code at [jmccrohan/pysolarmanv5 register_scan.py](https://github.com/jmccrohan/pysolarmanv5/blob/main/examples/register_scan.py)  doesn't detect anything.

Did you test 'holding' or 'input' registers?

Cheers,
Andrea

Pel1can111, 2025-03-03:
Firstly great work as I have been using the API up to now and having a local integration for my batteries is a huge improvement! 

A few things missing that I am trying to work out...

Battery power is provided via https://home.solarmanpv.com/. I can't find anything to suggest this is available locally so I guess this can just be worked out using voltage/current in HA.

Relay status is also on solarmanpv. It is either on or off but I'm struggling to find corresponding register.

Finally "status". This is either charging, discharging or idle and I also don't see a register for battery status. I guess this can also be worked out from the current (positive, negative or 0) but curious if there is more of a "status" provided

I plan on monitoring using the python script over time and see if anything jumps out.

Pel1can111, 2025-03-03:
Is there any reason not to join the requests for:

  # BMS Status
  - start: 5120
    end: 5135
    mb_functioncode: 0x03

  # Cell/Module Max/Min Temperature and Voltage
  - start: 5136
    end: 5151
    mb_functioncode: 0x03

  # Total Charge, Discharge
  - start: 5152
    end: 5167
    mb_functioncode: 0x03

Could they not just be:

  # BMS Status/Cell/Module Max/Min Temperature and Voltage/Total Charge, Discharge
  - start: 5120
    end: 5167
    mb_functioncode: 0x03