StephanJoubert/home_assistant_solarman #537: All sensors show as unknown

dmartin794, 2024-03-14:
Solarman integration via HACS was working fine until last night. From what I can tell nothing changed, except the laptop that I am running HA on updated and rebooted. 
Can still connect to Inverter via IP, and Inverter is still connected to Solarman Smart and pushing data. 
When looking at the logs in HA I have the following error:

NoSocketAvailableError: No socket available
(SyncWorker_4) [custom_components.solarman.solarman] Querying [3 - 112] failed with exception [NoSocketAvailableError: No socket available]
On the entity I have an alert:
This entity is no longer being provided by the solarman integration. If the entity is no longer in use, delete it in settings.
Using HA 12.1 with 2024.3.0 Core.

giedriuskilcauskas, 2024-03-18:
Most likely docker was updated. There were some other issues opened and workaround was to downgrade it and lock.

parkflyer, 2024-03-22:
Hi!
I have the same problem (same errors as above) around every 2-4 days (since several months now). Restarting HA solves the issue, but is no real solution. 
Is there a possibility to relaod the integration without restarting HA?
I'm using a Deye 12K, all components re in their current version.

darkman82, 2024-03-29:
Hi, 
open SSH Terminal and run: `netstat -anp | grep 8899`

How many connections do you see?

parkflyer, 2024-03-30:
Caro Guiseppe!

 

“netstat -anp | grep 8899” returns nothing (entered in HA Terminal).

Solarman currently runs without problems (last “hang” was 2 days ago).

 

netstat –anp returns:

 

Con cordiali saluti,

Ewald

darkman82, 2024-03-30:
Hi,
you should run netstat when the system hang, however you should always see at least one connection. Are you using the sticker logger on the standard port (8899) ?

parkflyer, 2024-03-31:
Dear Guiseppe!

 

Yes, I’m using the WLAN stick on the Deye on port 8899:

 

I can connect via the web interface via the IP address (also when Solarman is not responding any more).

 

Note that I also have a Wifi Modbus adapter (https://github.com/klatremis/esphome-for-deye) connected to my Deye which works flawlessly.

But I got the HA Energy Dashboard only working with Solarman.

I’m using DuckDNS as DynDNS to access my HA / Raspberry Pi (not sure of this is important).

 

I also do not have a connection to any Remote Server for the Deye:

 

Anyway for most of the time it works OK. 

 

Sincerely,

Ewald

darkman82, 2024-03-31:
Have a look at #523

parkflyer, 2024-04-11:
Thanks, I‘ve applied the fix you mentioned and it is now working  for longer than a week, no problems since then!