StephanJoubert/home_assistant_solarman #539: Works well with a master, not with a slave

rafal98, 2024-03-19:
Hi Stephan,

I add in ha my first inverter and it works well :-) Then a couple of days after I configure another inverter as a slave (with parallel option) and add it too to ha. Configuration seems ok in the plugin, but no info are reported on this second device. 

So the log show some interesting errors. Most interesting log is " failed with exception [V5FrameError: V5 frame contains **incorrect data logger serial number**"
But I'm sure that my logger SN (not inverter SN) is correct since with another tool I grab data
('192.168.2.142', 8899) => this is the IP of the second inverter
2907701578 is the correct SN of the second inverter.

Notes: 
- I try to set modbus to 1 then 2, it change nothing (master is 1, and slave is 2). I also try to point to the IP of the master, no change
- logger firmware version: LSW3_15_FFFF_1.0.9E
- inverters are  SUN-6K-SG03LP1-EU

```
2024-03-19 16:40:47.825 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Starting to query for [3] ranges...
2024-03-19 16:40:47.825 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [3 - 112]...
2024-03-19 16:40:47.825 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 11 00 71 95 41 ad 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 6e 34 26 42 15
2024-03-19 16:40:47.826 DEBUG (SyncWorker_16) [custom_components.solarman.solarman] Starting to query for [3] ranges...
2024-03-19 16:40:47.827 DEBUG (SyncWorker_16) [custom_components.solarman.solarman] Querying [3 - 112]...
2024-03-19 16:40:47.827 INFO (SyncWorker_16) [custom_components.solarman.solarman] Connecting to solarman data logger 192.168.2.142:8899
2024-03-19 16:40:47.833 DEBUG (SyncWorker_16) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=78, family=2, type=1, proto=6, laddr=('192.168.2.25', 38660), raddr=('192.168.2.142', 8899)>
2024-03-19 16:40:47.833 DEBUG (SyncWorker_16) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 95 00 a4 de ce 89 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 6e 34 26 ab 15
2024-03-19 16:40:48.105 DEBUG (SyncWorker_16) [custom_components.solarman.solarman] RECD: a5 10 00 10 15 95 68 4a 01 50 ad 02 01 73 39 04 00 8f 0f 00 00 8b 78 f5 65 06 00 2e 15
2024-03-19 16:40:48.105 WARNING (SyncWorker_16) [custom_components.solarman.solarman] Querying [3 - 112] failed with exception [V5FrameError: V5 frame contains incorrect data logger serial number]
2024-03-19 16:40:48.106 INFO (SyncWorker_16) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192.168.2.142:8899
2024-03-19 16:40:48.129 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] RECD: a5 ef 00 10 15 11 93 71 95 41 ad 02 01 62 83 24 00 e9 26 00 00 9c 2e d5 65 01 03 dc 32 33 31 32 30 33 36 30 31 35 00 00 81 02 00 00 15 15 00 00 43 85 c3 69 00 00 ea 60 00 00 02 01 00 01 00 ff 00 0a 18 03 13 10 27 15 00 01 17 70 05 14 00 00 00 00 00 00 00 04 00 00 00 00 00 00 07 d0 1f 72 00 00 00 00 03 e8 ff ff 00 00 00 00 00 01 00 00 00 01 00 00 00 00 00 00 00 00 ea 60 00 00 00 00 19 64 00 01 00 00 00 0c 00 01 00 01 00 02 00 66 00 fa 00 00 0a b1 00 00 09 ec 08 f6 07 78 0c bc 00 00 00 5f 00 0f 03 3a 00 00 02 5a 00 00 00 01 00 00 07 22 00 00 00 00 01 f7 00 00 00 00 00 68 10 28 00 00 0e fa 00 00 00 00 06 03 05 7c 00 00 00 01 07 d0 00 00 0d 59 00 00 0a 02 00 00 0a 03 00 00 00 00 00 00 00 00 00 00 00 00 01 2c 00 c8 0b 0d 00 05 0b 30 00 05 d9 8b a1 15
2024-03-19 16:40:48.131 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [3 - 112] succeeded
2024-03-19 16:40:48.131 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [150 - 249]...
2024-03-19 16:40:48.131 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 12 00 71 95 41 ad 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 96 00 64 a4 0d 23 15
2024-03-19 16:40:48.409 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] RECD: a5 db 00 10 15 12 94 71 95 41 ad 02 01 63 83 24 00 e9 26 00 00 9c 2e d5 65 01 03 c8 00 18 00 00 00 18 00 00 08 fd 00 00 08 fd 09 1e 00 00 00 00 00 00 00 00 00 00 00 00 00 5a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a8 00 00 00 a8 00 a8 00 00 00 a8 00 a8 00 00 00 65 04 d9 14 71 00 64 00 a8 00 87 00 98 00 00 00 00 ff f6 ff ec 13 88 13 88 00 00 00 10 00 00 00 00 00 00 00 00 00 01 16 80 16 80 15 90 01 2c 11 94 00 14 00 5a 00 00 00 00 00 50 00 64 00 00 00 01 00 00 00 08 03 de 00 05 00 32 00 0f 11 f8 14 50 12 8e 00 f0 00 00 13 24 00 1e 00 28 13 24 00 1e 00 28 00 00 00 00 00 00 00 00 00 00 13 ec 00 5f 15 18 00 64 00 00 00 00 00 00 00 01 00 02 01 f4 00 00 00 00 00 ff 00 00 e4 e8 40 15
2024-03-19 16:40:48.410 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [150 - 249] succeeded
2024-03-19 16:40:48.410 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [250 - 279]...
2024-03-19 16:40:48.411 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 13 00 71 95 41 ad 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 fa 00 1e e5 f3 69 15
2024-03-19 16:40:48.549 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] RECD: a5 4f 00 10 15 13 95 71 95 41 ad 02 01 63 83 24 00 e9 26 00 00 9c 2e d5 65 01 03 3c 00 00 00 c8 02 58 04 ce 05 fa 08 34 13 88 13 88 17 70 17 70 17 70 17 70 13 24 13 24 13 24 13 24 13 24 13 24 00 14 00 28 00 14 00 28 00 14 00 14 00 04 00 05 00 04 00 05 00 04 00 04 ef fb df 15
2024-03-19 16:40:48.550 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [250 - 279] succeeded
2024-03-19 16:40:48.550 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] All queries succeeded, exposing updated values.
2024-03-19 16:40:48.606 WARNING (SyncWorker_16) [custom_components.solarman.solarman] Querying [3 - 112] failed, [1] retry attempts left
2024-03-19 16:40:48.606 INFO (SyncWorker_16) [custom_components.solarman.solarman] Connecting to solarman data logger 192.168.2.142:8899
2024-03-19 16:40:48.612 DEBUG (SyncWorker_16) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=78, family=2, type=1, proto=6, laddr=('192.168.2.25', 38672), raddr=('192.168.2.142', 8899)>
2024-03-19 16:40:48.612 DEBUG (SyncWorker_16) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 b4 00 a4 de ce 89 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 6e 34 26 ca 15
2024-03-19 16:40:48.652 DEBUG (SyncWorker_16) [custom_components.solarman.solarman] RECD: a5 10 00 10 15 b4 69 4a 01 50 ad 02 01 74 39 04 00 90 0f 00 00 8b 78 f5 65 06 00 50 15
2024-03-19 16:40:48.653 WARNING (SyncWorker_16) [custom_components.solarman.solarman] Querying [3 - 112] failed with exception [V5FrameError: V5 frame contains incorrect data logger serial number]
2024-03-19 16:40:48.653 INFO (SyncWorker_16) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192.168.2.142:8899
2024-03-19 16:40:49.153 WARNING (SyncWorker_16) [custom_components.solarman.solarman] Querying [3 - 112] failed, [0] retry attempts left
2024-03-19 16:40:49.153 WARNING (SyncWorker_16) [custom_components.solarman.solarman] Querying registers [3 - 112] failed, aborting.
2024-03-19 16:40:49.154 DEBUG (SyncWorker_16) [custom_components.solarman.sensor] No value recorded for PV1 Power
2024-03-19 16:40:49.155 DEBUG (SyncWorker_16) [custom_components.solarman.sensor] No value recorded for PV2 Power
2024-03-19 16:40:49.157 DEBUG (SyncWorker_25) [custom_components.solarman.sensor] No value recorded for PV1 Voltage
2024-03-19 16:40:49.158 DEBUG (SyncWorker_56) [custom_components.solarman.sensor] No value recorded for PV2 Voltage
2024-03-19 16:40:49.159 DEBUG (SyncWorker_26) [custom_components.solarman.sensor] No value recorded for PV1 Current
[...]
```
in this log, we can see the config of the 2nd device:

2024-03-19 16:42:53.205 DEBUG (MainThread) [custom_components.solarman.config_flow] config_flow.py:step_user_data_schema: {'name': 'deye2', 'inverter_host': '192.168.2.142', 'inverter_serial': 2907701578, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_hybrid.yaml'}

Regards,

Laurent

rafal98, 2024-03-20:
I think the issue is linked to a connect optimization that keep the last connection, but it can't work if you have several inverters

My change:

```
    def connect_to_server(self):
        if self._modbus **and self.last_serial == self._serial:**
            return self._modbus
        log.info(f"Connecting to solarman data logger {self._host}:{self._port} (serial: {self._serial}, modbus: {self._mb_slaveid})")
        self._modbus = PySolarmanV5(self._host, self._serial, port=self._port, mb_slave_id=self._mb_slaveid, logger=log, auto_reconnect=True, socket_timeout=15)
        **self.last_serial = self._serial**
```

Now it works :-)

Elijen, 2024-03-28:
@rafal98 Does the change to `connect_to_server` fix the issue of unknown values for the second inverter? Looking at the code it seems that each integration has their own `Inverter` instance so I don't understand how the connections can get mixed up.

rafal98, 2024-03-28:
Hi @Elijen, it's perhaps a side effect, I don't know, but after my quick fix, it works like a charm: I have all entities with updated values

Elijen, 2024-03-28:
I have noticed my error message was different:
```
V5FrameError: V5 frame does not contain a valid Modbus RTU frame
```
After downgrading from `1.5.1` to version `1.4.3` I started getting a parsing error that I solved by changing validation rules in the inverter config file. For some reason slave inverter is reporting battery temperature as `-100` and the integration was throwing all data away because of it.

ornulf, 2024-04-03:
> I think the issue is linked to a connect optimization that keep the last connection, but it can't work if you have several inverters
> 
> My change was to keep the last serial used. If it's the same, so ok use the current connection. Else it's probably necessary to close the current connection before opening a new one, not sure about that
> 
> ```
>     def connect_to_server(self):
>         if self._modbus and self.last_serial == self._serial:
>             return self._modbus
>         log.info(f"Connecting to solarman data logger {self._host}:{self._port} (serial: {self._serial}, modbus: {self._mb_slaveid})")
>         self._modbus = PySolarmanV5(self._host, self._serial, port=self._port, mb_slave_id=self._mb_slaveid, logger=log, auto_reconnect=True, socket_timeout=15)
>         self.last_serial = self._serial
> ```
> 
> Now it works :-)

Hi @rafal98 

I have the same problems as you with my second inverter Deye SUN-10K-SG04LP3-EU. My master inverter is showing the correct values but my second configured logger/inverter is showing all values as only unknown. Could you please elaborate a little further how you solved the proplem. I'm a complete newibe so have patience with my ignorance.

rafal98, 2024-04-03:
Hi Ornulf,

I suggest you install the "Studio code server" add-on, and openit then browse folder "CONFIG/custom_components/solarman/" then open/edit "solarman.py" then replace the "connect_to_server" function by my modified version (on previous post). You will see that the difference is small (I only add self.last_serial variable)

Have fun !

ornulf, 2024-04-03:
Cheers. I will test that👍Skickat från min iPhone3 apr.

Klagio, 2024-09-15:
this is the original, I am trying the changed code
```

    def connect_to_server(self):
        if self._modbus:
            return self._modbus
        log.info(f"Connecting to solarman data logger {self._host}:{self._port}")
        self._modbus = PySolarmanV5(self._host, self._serial, port=self._port, mb_slave_id=self._mb_slaveid, logger=log, auto_reconnect=True, socket_timeout=15)
```

Klagio, 2024-09-15:
> Laurent

Hi, where can I see the LOG?

I also can see my Master values and Slave are all Unknown, tried to modify also the solarman.py, but same resul

rafal98, 2024-09-16:
Hi Klagio,

You need to Enable debug logging, then play with the plugin, restart or
whatever, then click on "Disable debug logging" when you have done enough
test, it will download the full debug logs (with others devices too)

[image: image.png]