StephanJoubert/home_assistant_solarman #542: Logger starting 29xxxxxx all values are unknown

alessiopetrozziello, 2024-03-21:
Hello,

my logger serial number starts with 29xxxxxxx, and when I configure solarman through HA, i get all unknown, can I use this repo integration or should i use a different one?

Thanks

CezaryZwierzchowski, 2024-04-08:
Hello, 

I have similar logger 29xxxx and the same problem. All parameters are unknown. When I open web GUI for logger I have to type the user name and password. The password I know but what username I should be used for that?

VAC91, 2024-04-08:
Login and password should be: admin.
I have logger serial 29xxxx too and I'm using 1.4.3 version of this integration. Works well for me.

CezaryZwierzchowski, 2024-04-09:
Hi, I spent some more time and now it's working fine. I also updated the HA core and OS to the latest version. Thanks for replay!

deye_sg04lp3.yaml with Deye 10K - SG04LP3

PV-Joe, 2024-04-10:
The Solarman Integration works perfect and stable since more than a year with:

Deye 8K SG04LP3-EU
LSW3 Stick-Logger
Solarman 1.4.3
Update interval: 20 seconds

THX to Stephan for his great work!

PV-Joe, 2024-04-10:
> Update interval 20 seconds? How can i change this value to 20sec too? My logger sends data every 1 minute.

Here you go:

[image]

MIN_TIME_BETWEEN_UPDATES = timedelta (seconds=20)

solar4ever, 2024-05-18:
@alessiopetrozziello  I would be indebted, if anyone could help me with [this issue](https://github.com/jmccrohan/pysolarmanv5/issues/56#issue-2302648839).