StephanJoubert/home_assistant_solarman #545: Afore Inverter

olovfr, 2024-03-27:
My Serial number starts with 27. I manage to connect to my Afore Inverter using the definition for Afore inverters. But there is no menu item on the inverter to change to modbus like it suggests in the definition. All values are wrong that I receive.
Is there anything I can do to make the connection working.

ketoha, 2024-05-12:
What afore inverter do you have? And what version of solarman are you using?

olovfr, 2024-05-15:
My inverter is bnt005ktl and solarman version is 1.4.3
I have been able to connect directly by modbus rtu also but i recieve the same wrong values. I found in the inverter to set communication port 2 to modbus but that didnt help. What menu changes rs485 to modbus?

Bielik1984, 2026-09-13:
Today I finally managed to connect my AFORE BNT005KTL to Home Assistant locally using an LSW3 logger and the StephanJoubert Solarman integration, after struggling with this for four years!

My inverter has no menu option to switch from RS485 to Modbus, but local communication works without changing that setting.

Save the configuration below as `custom_parameters.yaml` in:

`/config/custom_components/solarman/inverter_definitions/`

Then select `custom_parameters.yaml` in the Solarman integration settings and restart Home Assistant.

This profile uses the AFORE T6 register map. Phase voltages and grid frequency have been confirmed against live readings. Further validation is still in progress, especially for the energy counters, so please compare those with your inverter readings before using them in the Energy dashboard.

```yaml
# AFORE BNT005KTL T6 — read-only. Numeric sensors require class and uom, even when null.
# Verified live: 507-509, 513. Other ranges require validation.
# 32-bit lists: low word first. Upstream rule 2 has a one-unit negative decoding error.
# Energy: verify 0.1 kWh scaling against inverter before using statistics.
requests:
  - start: 0
    end: 53
    mb_functioncode: 4
  - start: 500
    end: 528
    mb_functioncode: 4
  - start: 553
    end: 560
    mb_functioncode: 4
  - start: 1000
    end: 1015
    mb_functioncode: 4
  - start: 2500
    end: 2500
    mb_functioncode: 4
  - start: 2514
    end: 2515
    mb_functioncode: 4
  - start: 3000
    end: 3027
    mb_functioncode: 4
parameters:
  - group: solar
    items:
      - name: Afore Model
        rule: 5
        registers:
          - 0
          - 1
          - 2
          - 3
          - 4
          - 5
        isstr: true
      - name: Afore Serial number
        rule: 5
        registers:
          - 41
          - 42
          - 43
          - 44
          - 45
          - 46
          - 47
          - 48
        isstr: true
      - name: Afore Rated power
        rule: 1
        registers:
          - 8
          - 7
        uom: W
[... 436 lines cut ...]
      - name: Afore Permanent error mask
        rule: 1
        registers:
          - 3023
          - 3022
        scale: 1
        class: null
        uom: null
      - name: Afore Internal warning mask
        rule: 1
        registers:
          - 3025
          - 3024
        scale: 1
        class: null
        uom: null
      - name: Afore External warning mask
        rule: 1
        registers:
          - 3027
          - 3026
        scale: 1
        class: null
        uom: null
```