StephanJoubert/home_assistant_solarman #546: integrazione solarman 27...... con INVERTER INVT BD6KTL

Batcav, 2024-03-27:
Ciao ho provato tutte le configurazioni possibili ma nessuna sembra funzionare, riporta sempre tutti i sensori come sconosciuti o non disponibile. Prima di provare in API , chi sarebbe cosi gentile da darmi una mano e provare a configurare questo inverter e condividerlo? Logger solarman 27......
Grazie

81ondo, 2024-04-10:
Write and email to the solarman support. 
In 1-2 weeks they can provide you the modbus register document for your inverter, with that you'll be able to modify an existing configuration file to adapt it for your inverter

Batcav, 2024-04-11:
grazie per la risposta, allora se intendi i  codici API quelli li ho già,
ho inoltrato email a solarman che già il giorno dopo mi ha risposto
fornendomi i codici, la questione è che non ho capito come e dove inserirli.

xaras2, 2025-04-21:
> grazie per la risposta, allora se intendi i  codici API quelli li ho già,
> ho inoltrato email a solarman che già il giorno dopo mi ha risposto
> fornendomi i codici, la questione è che non ho capito come e dove inserirli.
> 
> Il Mer 10 Apr 2024, 20:14 81ondo ***@***.***> ha scritto:
> […](#)

Ti andrebbe di condividerli, ho lo stesso inverter (BD6KTL-RL1), vorrei provare ad integrarlo con Home Assistant.
Grazie