StephanJoubert/home_assistant_solarman #547: Logger 27XXXXXXXX with ZCS Hybrid Inverter (Model NO: AZZURRO 3PH HYD6000 ZSS)

al31c0, 2024-03-28:
Hi everyone!

I'm trying to connect my Solar PV and Battery System composed by:
1. ZCS Hybrid Inverter (Model NO: AZZURRO 3PH HYD6000 ZSS)
2. logger with serial number **27XXXXXXXX** and Firmware version **LSW3_15_270A_1.68**

I configured the Solarman integration with the file **zcs_blue-ktl-v3.yaml** and the system seems to work partially: the integration exposes 76 entities but only some are available (see attached photo).

I have already tried to downgrade Solarman to 1.4.3 but the result is the same: some entities are available, others are not. 

I therefore do not think it is a problem with the logger but with the .yaml configuration file. Can anyone give me some pointers on how to reconstruct it for my inverter?

Thank you for your support

[image]

[image]

darkman82, 2024-03-29:
Try sofar_hyd3k-6k-es.yaml

solar4ever, 2024-05-18:
@al31c0 Like you I need the help of community at this time. Since you seem to have similar setup to mine (i.e., Azzurro and Solarman Logger), I would greatly appreciate it, if could you please check [my comment on pysolarmanv5](https://github.com/jmccrohan/pysolarmanv5/issues/56#issue-2302648839), and let me know if you noticed similar behaviour in your solarman smart acount etc.?