StephanJoubert/home_assistant_solarman #548: SOLARMAN INTEGRATION BRIKEN

mattomu22, 2024-03-29:
Very simple. After last HA update the integration stopped working. Sensors unknown. The inverter and the wifi logger are working normally. I can read values with inverter app. 
Thank you.

Dragonstout, 2024-03-29:
I also can confirm this. No Data get received with:
 Core 2024.4.0b1
Supervisor 2024.03.1
Operating System 12.2.rc1 

Version

mattomu22, 2024-03-29:
I tried to remove a reinstall but no change. Any ideas? Thanks

toivio, 2024-03-30:
The same

PV-Joe, 2024-03-30:
Just updated to:

[image]

… and everything is working fine ✅

mattomu22, 2024-03-30:
I tried to load back up but didn't go well. I will wait for possible fix in my case.

toivio, 2024-03-30:
How can i do downgrade?
[image]

PV-Joe, 2024-03-30:
> How can i do downgrade? ![Screenshot_20240330-115254](https://private-user-images.githubusercontent.com/17320070/318188311-0ea23e22-12c3-40f5-a19b-9f32865edb66.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE4MTA5ODksIm5iZiI6MTcxMTgxMDY4OSwicGF0aCI6Ii8xNzMyMDA3MC8zMTgxODgzMTEtMGVhMjNlMjItMTJjMy00MGY1LWExOWItOWYzMjg2NWVkYjY2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzMwVDE0NTgwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMzODc3Njk0NTVlMTkxZTkxNjA2MmM1ZTA2YzllNzMwYmRmMjFiZjk4NGQzMGFkMDViOWFkYzUyNDljNWQyMGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.SY_FPto8kHefXcL3aLb-zjbFWiEFy6ve5eI3CYg6sYU)

Restore from backup?
Beta releases can be faulty.

toivio, 2024-03-31:
> Restore from backup? Beta releases can be faulty.

I have restore, manual config is ok, but all sensors have an error stare.

jpameli, 2024-03-31:
Same here I am on:
Core 2024.3.3
Supervisor 2024.03.1

toivio, 2024-03-31:
> Same here I am on: Core 2024.3.3 Supervisor 2024.03.1

Same, that's meen Ok or not Ok.

jpameli, 2024-03-31:
Its broken, sorry!

toivio, 2024-03-31:
> Its broken, sorry! Maybe its a Solarman issue? I can't even see my devices at https://home.solarmanpv.com . They show offline but they are not.

https://home.solarmanpv.com its Ok for me, but solarman is working on lokal IP add.

jpameli, 2024-03-31:
You are right, its too late here and they went offline because of low sun.

mattomu22, 2024-03-31:
Personally I will not revert because I did a lots of things in the last few days.
Can someone tell me if you are going to fix it? I pay annual subscription to Home Assistant. Thank you.

maxsiindonesia, 2024-04-01:
Indonesia is also the same, it doesn't work after updating

herijaya, 2024-04-01:
> Indonesia is also the same, it doesn't work after updating

same

liviui, 2024-04-01:
The same here, I've also updated to 12.2: Entity unavailable

schasan, 2024-04-02:
I do not think "broken" is the correct wording.
Deye seem to have dropped their previous old style modbus in favour of "AT+..." commands, which this integration does not (yet) support.
Try something like the following to receive a steady stream of data:

import socket
import binascii

def dump(binary, size=2, sep=' '):
  hexstr = binascii.hexlify(binary)
  hexstr = hexstr.decode('ascii')
  return hexstr.upper()

s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.connect(("192.168.8.16", 8899))

while True:
    buffer = s.recv(80)
    print(len(buffer))
    print(dump(buffer))
    print(buffer)

mattomu22, 2024-04-02:
My system is a ZCS Azzurro equivalent to sofar inverter and I was sure I am
taking the data directly from the inverter via wifi. May be is on of those
things that at the end was my fault and I have addes something that goes in
conflict. Boh!%@£#!% I will try to disconnect the wifi dongle and try all
again.
This is for me.

Xabrim, 2024-04-03:
I solved the problem by downgrading to 1.4.3. Delete all downloaded Hacs files (hacs-> Solarman-> Top right corner-> Delete). Delete the Integration in HA. Restart. Download 1.4.3. Restart. Configure the Integration. Restart again and the values changed from unknown to solid values.

Hassfabi, 2024-04-03:
> I solved the problem by downgrading to 1.4.3. Delete all downloaded Hacs files (hacs-> Solarman-> Top right corner-> Delete). Delete the Integration in HA. Restart. Download 1.4.3. Restart. Configure the Integration. Restart again and the values changed from unknown to solid values.

Worked for me at least for one inverter. I have set up two Deye 600 in Solarman. One with SN 39xxx, which was also the first inverter I set up in Solarman and this one worked (again) after downgrading to 1.4.3.
The other one with SN 41xxx is unfortunately still not working.
Any clue? Maybe the Integration can't handle two loggers/inverter?

xwitte, 2024-04-08:
Same for me after last HA update. 

Neither downgrade to 1.4.3. nor solution in pull request #550 worked for me. Still no values for sensor data. On version 1.51 I saw that the graphs sometimes showed single values spread over the whole day. But values could not be gathered regularly.

2024-04-08 11:48:06.202 WARNING (MainThread) [homeassistant.components.sensor] Updating solarman sensor took longer than the scheduled update interval 0:00:30
2024-04-08 11:48:12.365 WARNING (SyncWorker_20) [custom_components.solarman.solarman] Querying registers [3 - 112] failed, aborting.
2024-04-08 11:49:16.205 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.pv1_stand_pv1_power is taking over 10 seconds
2024-04-08 11:49:36.206 WARNING (MainThread) [homeassistant.components.sensor] Updating solarman sensor took longer than the scheduled update interval 0:00:30
2024-04-08 11:49:42.263 WARNING (SyncWorker_47) [custom_components.solarman.solarman] Querying registers [3 - 112] failed, aborting.

My specs:
Inverter: Deye SUN800G3-EU-230
Firmware version: MW3_16U_5406_1.57

Home Assistant

Core: 2024.4.1
Supervisor: 2024.03.1
Operating System: 12.1
Frontend: 20240404.1

Xabrim, 2024-04-10:
Check your Version again. Sometimes it auto updates/loads the wrong version. Delete Hacs file an Integration and try a full HA reboot before installing Version 1.4.3.

xwitte, 2024-04-11:
> Check your Version again. Sometimes it auto updates/loads the wrong version. I had this problem with a different system. Delete Hacs file and the integration and try a **full HA reboot** before installing Version 1.4.3. Tell me if it works for you.

You are right. I repeated the procedure and fully reloaded Home Assistant and then 1.4.3 was working as expected and shows collected sensor values. 

Thx @Xabrim 

See also https://github.com/StephanJoubert/home_assistant_solarman/issues/467#issuecomment-1868973696

erxbout, 2024-04-11:
Hi I also had this problem!

I noticed that after I updated HASS OS
Seems that it does not work anymore with HASS OS 12.2 but with 12.1 its good!

downgrade with this command
``` bash
ha os update --version 12.1
``` 

My working installation is now with versions:
core 2024.4.2
Supervisor 2024.04.0
OS 12.1
Frontend 20240404.1

jmccrohan, 2024-04-11:
This is an issue that has been introduced in HA OS 12.2 with the upgrade to Docker 25. It has impacted a number of containers (add-ons), including HA core itself.

The proper fix is to remain on HA OS 12.2 and to rebuild the core container:

`ha core rebuild`

No changes to the Solarman integration required.

erxbout, 2024-04-11:
Have tried that and can confirm that also works!

deebeo-blip, 2024-04-22:
Is this something that can be expected to be adressed in HA OS  in the future or will every future HA OS Update lead to the Solarman integration not working anymore?

wandido, 2024-07-17:
Trying to get Amensolar 8k to work on HA but he ting message simila

ayubn, 2024-10-31:
> This is an issue that has been introduced in HA OS 12.2 with the upgrade to Docker 25. It has impacted a number of containers (add-ons), including HA core itself.
>
> The proper fix is to remain on HA [...]

thanks alots its working now

francosax, 2025-04-06:
I have the same issue with las HA OS:

[image]

how to solve it?

Thank you!

pmenclik, 2025-06-17:
Hi I have same issue after core update from 2025.5.3 to 2025.6.1, OS 15.2. I have tried ha core rebuild, downgrade of core, even downgrade of os. No success. Solarman integration simply stopped working.  I am pretty sad, since I have quite some automations for this this integration  

Any other ideas, anybody can help?

ayubn, 2025-06-19:
> Hi I have same issue after core update from 2025.5.3 to 2025.6.1, OS 15.2. I have tried ha core rebuild, downgrade of core, even downgrade of os. No success. Solarman integration simply stopped [...]

ha core rebuild 
This command solved my issue