StephanJoubert/home_assistant_solarman #553: Solarman Integration with datalogger 39XXXXXXXXXX not working

nandogandra, 2024-04-07:
Hi there, does anyone have HA integration with data logger 39XXXXXXXX working ?
 
If yes, which version release have worked for you ?

nandogandra, 2024-04-07:
see below the error logs:

Registrador: homeassistant.util.package
Fonte: util/package.py:123
Ocorreu pela primeira vez: 15:37:05 (3 ocorrências)
Último registro: 15:37:53

Unable to install package pysolarmanv5: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f91176bd0>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f902b3e90>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f901b8920>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f901b8860>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f901b8bf0>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ ERROR: Could not find a version that satisfies the requirement pysolarmanv5 (from versions: none) ERROR: No matching distribution found for pysolarmanv5
Unable to install package pysolarmanv5: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f97f812e0>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f97e3c8c0>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f96bf4b60>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f96bf4ad0>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f96bf4e60>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ ERROR: Could not find a version that satisfies the requirement pysolarmanv5 (from versions: none) ERROR: No matching distribution found for pysolarmanv5
Unable to install package pysolarmanv5: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7faa2130b0>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fa9453c20>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fa9360590>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fa9360980>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fa9360d10>: Failed to establish a new connection: [Errno 101] Network unreachable')': /simple/pysolarmanv5/ ERROR: Could not find a version that satisfies the requirement pysolarmanv5 (from versions: none) ERROR: No matching distribution found for pysolarmanv5

Registrador: homeassistant.config
Fonte: config.py:1324
Ocorreu pela primeira vez: 15:37:53 (1 ocorrências)
Último registro: 15:37:53

Platform error: sensor - Requirements for solarman not found: ['pysolarmanv5'].

Registrador: homeassistant.setup
Fonte: setup.py:273
Ocorreu pela primeira vez: 15:37:53 (1 ocorrências)
Último registro: 15:37:53

Setup failed for custom integration 'solarman': Requirements for solarman not found: ['pysolarmanv5'].

bam91, 2024-04-19:
I got mine working. I think you have to use the **deye_2mppt.yaml** .

inigoml, 2024-04-21:
Same problem here with 4 MPTT G4 microinverter. Fixed by itself. Just wait until inverter is properly working, that is, is injecting energy to the network.