StephanJoubert/home_assistant_solarman #556: It does not work anymore

jolly12f, 2024-04-11:
after the latest Home Assistant update, the integration no longer reports any data

mirekma157, 2024-04-11:
to same umřelo

robert10055, 2024-04-11:
what you update OS? or Core?

jolly12f, 2024-04-11:
I'm currently restoring my system to the latest backup

frenk91, 2024-04-11:
is the same for me, looking at the logs i found this 
`2024-04-11 21:40:08.074 WARNING (SyncWorker_57) [custom_components.solarman.solarman] Querying registers [1028 - 1056] failed, aborting.
2024-04-11 21:40:38.069 WARNING (SyncWorker_15) [custom_components.solarman.solarman] Querying [1028 - 1056] failed with exception [OSError: [Errno 30] Read-only file system]
2024-04-11 21:40:38.071 WARNING (SyncWorker_15) [custom_components.solarman.solarman] Querying [1028 - 1056] failed, [1] retry attempts left
2024-04-11 21:40:38.074 WARNING (SyncWorker_15) [custom_components.solarman.solarman] Querying [1028 - 1056] failed with exception [OSError: [Errno 30] Read-only file system]
2024-04-11 21:40:38.075 WARNING (SyncWorker_15) [custom_components.solarman.solarman] Querying [1028 - 1056] failed, [0] retry attempts left
2024-04-11 21:40:38.076 WARNING (SyncWorker_15) [custom_components.solarman.solarman] Querying registers [1028 - 1056] failed, aborting.`

Marky0, 2024-04-11:
I found a fix here
[https://github.com/StephanJoubert/home_assistant_solarman/issues/557#issuecomment-2050437101](url)

Marky0, 2024-04-11:
A fix is here
[https://github.com/StephanJoubert/home_assistant_solarman/issues/557#issuecomment-2050437101](url)

jolly12f, 2024-04-11:
I can't even access the terminal

Marky0, 2024-04-11:
> I can't even access the terminal

Sorry, I can't help you there. If you have followed the explanation in link and the terminal is not working you've got bigger problems with HA