StephanJoubert/home_assistant_solarman #557: Solarman just stopped working

huwpower, 2024-04-11:


nicolacavallo84, 2024-04-11:
It stopped working for me too after I updated to OS version 12.2.  Are there already solutions?

NunzioGG, 2024-04-11:
From the HA terminal, type the command
has core rebuild
I have solved

Marky0, 2024-04-11:
@NunzioGG thanks. This worked for me too.
Install the Terminal & SSH Add-on if you don't already have it : 
[https://www.shellhacks.com/home-assistant-ssh-connection-using-terminal/](url)
Make sure you enable in sidebar to get easily get to the terminal window as described in the link

type "ha core rebuild"
Solarman will restart after a few minutes.

nicolacavallo84, 2024-04-11:
@NunzioGG thanks, with the command "ha core rebuild ha" it worked for me too

Bilo2110, 2024-04-11:
@Marky0 thx, after a few minutes it work

JakobHeubl, 2024-04-12:
the command "ha core rebuild" worked for me too

MartinN711, 2024-04-12:
"ha core rebuild" worked for me as well. Thanks @Marky0 @NunzioGG

lepereded, 2024-04-12:
"ha core rebuild" ok . Thanks a lot  @Marky0 @NunzioGG

margielm, 2024-04-12:
I just installed new version of HA - 2024.4.3 and that seemed to help as well. Without running the command.

Boerni-tech, 2024-04-12:
Neither the "ha core rebuild" nore the installation of 2024.4.3 fixed it for me. Still broken.

jouster1974, 2024-04-12:
came here to report the same issue, found the fix and implemented it, and can confirm now working fine on HASSOS12.2 and Core 2024.4.3

Thanks for the quick fix, first time Ive used that command

adiweps, 2024-04-12:
Worked for me too, thanx!!

Boerni-tech, 2024-04-13:
a second run and a reboot finally fixed it for me, as well.

81ondo, 2024-04-13:
Updated to version below and just rebooted and everything worked.

```
Core 2024.4.3
Supervisor 2024.04.0
Operating System 12.1
Frontend 20240404.2
```

Aredun, 2024-04-14:
Worked for me too, thanx!!

DominicStenzel, 2024-04-16:
ha core rebuild, worked for me too, thanks.

vishalkanhai, 2024-04-16:
Thank you for the resolution. ha core rebuild worked like a charm. much appreciated.

marianogi, 2024-04-19:
"ha core rebuild" worked but you must DOUBLE CHECK your **DEVICE** serial number, don't confuse it with _INVERTER_ serial number. It was driving me crazy.

Nailuj1, 2024-04-19:
ha core rebuild worked for me too

D3sp3holic, 2024-04-23:
Kiss it worked

philtechsl, 2024-04-23:
> ha core rebuild

It worked for me.

Thanks

UncleRiot, 2024-05-01:
After upgrading HA OS to 12.3.rc2 today and Solarman stopped working.

"ha core rebuild"
Worked for me also! Thanks a lot!

MarkusW71, 2024-05-02:
> "ha core rebuild" worked but you must DOUBLE CHECK your **DEVICE** serial number, don't confuse it with _INVERTER_ serial number. It was driving me crazy.

I kiss your eyes! That was the solution for me and maybe it could be better explained in the docs.