StephanJoubert/home_assistant_solarman #558: Definition for inverter ZCS Azzurro 1PH2200TL-V3

81ondo, 2024-04-12:
Filtered out entities not available on this inverter and optimized register ranges.

[ZCS_Azzurro_1PH2200TL-V3.zip](https://github.com/StephanJoubert/home_assistant_solarman/files/14956468/ZCS_Azzurro_1PH2200TL-V3.zip)

solar4ever, 2024-05-18:
@81ondo I would appreciate it, if you could give me some help on [this](https://github.com/jmccrohan/pysolarmanv5/issues/56#issue-2302648839). At least, since you seem to have an Azzurro Inverter, which provides the same parameters as mine, could you please let me know if you can see those parameters in your solarman smart app account?

solar4ever, 2024-05-19:
@81ondo I would also like to ask, how did  you find which register number corresponds to each parameter (in the` .yml` file inside the zip file you provided)? For instance, how did you find that "Inverter module temperature" has register of `[0x001B]`, and how `[0x001B]` translate to the register address that `pysolarmanv5` (which is used behind the scenes by this repo) accepts (see their example [here](https://github.com/jmccrohan/pysolarmanv5/blob/b22e5da918a35a50e262c95fe77095a61b9f69a8/examples/client_example.py#L17))?

double1968, 2024-05-20:
> Filtered out entities not available on this inverter and optimized register ranges.
> 
> [ZCS_Azzurro_1PH2200TL-V3.zip](https://github.com/StephanJoubert/home_assistant_solarman/files/14956468/ZCS_Azzurro_1PH2200TL-V3.zip)

Many Thanks