StephanJoubert/home_assistant_solarman #561: Problem connecting to Sofar ME3000SP

stephenchilds76, 2024-04-16:
Trying to connect via local IP have configured for port 8899 and ID 1.

Device serial number 4050986569

Firmware version MW_08_512_0501_1.94

Following is in the log:

2024-04-16 15:24:03.808 INFO (SyncWorker_10) [custom_components.solarman.solarman] Connecting to solarman data logger 192.168.1.10:8899
2024-04-16 15:24:03.862 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=21, family=2, type=1, proto=6, laddr=('192.168.1.52', 48718), raddr=('192.168.1.10', 8899)>
2024-04-16 15:24:03.863 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 c7 00 49 26 75 f1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 00 00 46 c5 80 9b 15
2024-04-16 15:24:05.057 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] RECD: a5 10 00 10 15 c7 83 49 26 75 f1 02 01 10 0f 90 01 a3 0a 00 00 f4 7a 8e 64 05 00 19 15
2024-04-16 15:24:05.058 WARNING (SyncWorker_10) [custom_components.solarman.solarman] Querying [512 - 581] failed with exception [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
2024-04-16 15:24:05.058 INFO (SyncWorker_10) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192.168.1.10:8899

stephenchilds76, 2024-04-16:
PS using wifikit template

StannekeS, 2024-05-07:
Same error here since last night. Did not run any updates.