StephanJoubert/home_assistant_solarman #568: Cannot get my 27XXXXXXXX series logger to work

cr07rupak, 2024-04-22:
Hi everyone!

I'm trying to connect my On-Grid Solar PV system composed by:

Deye Grid connected PV Inverter (SUN-3K_-G03)
logger with serial number 27XXXXXXXX and Firmware version LSW3_15_FFFF_1.0.9E
I am a newbie so I dont know what I'm doing wrong. I followed all the steps in the guide but cannot get it working

matoelorriaga, 2024-04-24:
similar situation, my module version is LSW3_15_270A_1.68 and my serial number is 27XXXXXXXX

WimDeceuninck, 2024-04-27:
Hi, 

Similar sitution here.

My module is a Solis S3-WIFI-ST datalogger with serial number 51215146. 
I do not know its firmware version.
How can I check the firmware version?
I can ping the device but I cannot log into it.

My inverter is a Solis Rhi-3.6k-48es-5g

I use Solarman version 1.4.3 with solis_hybrid.yaml (I downgraded from version 1.5.1 as I saw a post where some succeeded to resolve their connection issues with Solarman version 1.4.3 ). 

I am also a newbie and I don't know what I'm doing wrong.
The integration installation via HACS went well, but the connection is 'disconneccted' and the entities are 'unknown'.

I hope anyone can help me out...

Kind regards,
Wim

jmccrohan, 2024-04-28:
> My module is a Solis S3-WIFI-ST datalogger with serial number 51215146. I do not know its firmware version. How can I check the firmware version? I can ping the device but I cannot log into it.

Unfortunately the Solis S3-WIFI-ST is not a Solarman based data logger and this integration will not work for you. I suggest you look at [solis-sensor](https://github.com/hultenvp/solis-sensor) (cloud polling) or learn about [RS-485](https://github.com/alienatedsec/solis-ha-modbus-cloud).

WimDeceuninck, 2024-04-29:
Hi Jonathan,

Thank you for your confirmation. 

Indeed, in the mean time I found https://github.com/alienatedsec/solis-ha-modbus-cloud that also gives a nice overview of the differtent types of Solis dataloggers and their capabilities.

I have ordered a new DSL-W datalogger, I'll give it a try if it gets delivered (not sure if the supplier can deliver, nor if they will deliver the DSL-W as in @alienatedsec 's overview).

I'll keep you posted. 

Kind regards,
Wim

PCB0815, 2024-06-21:
@WimDeceuninck how have you downgraded? It does not let me download an older version?

WimDeceuninck, 2024-06-22:
Hi Stephan,

 

Downgrade is easy via HACS.

 

Select the integration.

Go to ‘redownload’.

And chose the version you want to use.

 

Kind regards,

Wim

 

This email has been checked for viruses by Avast antivirus software.
www.avast.com