StephanJoubert/home_assistant_solarman #569: Solarman integration looses the connection to logger

juergen-miessmer, 2024-04-23:
Since last update the solarman integration looses the connection to the loggers frequently an never regains it.
This happens randomly several times a day. After restarting HA it coonects again to the loggers.

ChrisJa176, 2024-04-23:
I have the same problem. no connection. not even after the restart.

philtechsl, 2024-04-23:
This worked for me.

See Issue : Solarman just stopped working #557

ChrisJa176, 2024-04-23:
Command "ha core rebuild" helps.

juergen-miessmer, 2024-04-24:
> Command "ha core rebuild" helps.

No, had allready done this twice,  issue still remains.

5h4mr0Ck, 2024-04-25:
Happened to me today where HA suddenly lost connection and the app was still active.  I thought that some how the inverter had restarted and changed it's IP address away from it's fixed value.  After a couple of hours (as I was at work and couldn't connect to the inverter or router remotely) I thought I'd try and ping, from the terminal, a previous IP address that the inverter has been using. No luck.  So I tried the fixed IP address and it pinged back.  I then checked my solarman integration and it was getting values again.  It seemed that the ping might have forced things back for the integration.

I then added an alert if it drops it again (if the battery % is unknown).

I also added the inverter IP address to the PING integration, just in case the ping did get it restarted.  If it did, then the regular check from PING should keep it working.

juergy58, 2024-04-26:
I've have had the same problem. Restart has solved it.

philtechsl, 2024-04-26:
I don´t have that kind of problem. Mine works fine after core rebuild.

juergen-miessmer, 2024-04-29:
No, ping dosn't  help to reconnect.
I also had a permanet ping running for two days now:  No change, connection gets still frequently lost and no reconnect.

> Happened to me today where HA suddenly lost connection and the app was still active. I thought that some how the inverter had restarted and changed it's IP address away from it's fixed value. After [...]

5h4mr0Ck, 2024-04-29:
> No, ping dosn't help to reconnect. I also had a permanet ping running for two days now: No change, connection gets still frequently lost and no reconnect. I need to restart manually every time.
> 
> > Happened to me today where HA suddenly lost connection and the app was still active. I thought that some how the inverter had restarted and changed it's IP address away from it's fixed value. After a couple of hours (as I was at work and couldn't connect to the inverter or router remotely) I thought I'd try and ping, from the terminal, a previous IP address that the inverter has been using. No luck. So I tried the fixed IP address and it pinged back. I then checked my solarman integration and it was getting values again. It seemed that the ping might have forced things back for the integration.
> > I then added an alert if it drops it again (if the battery % is unknown).
> > I also added the inverter IP address to the PING integration, just in case the ping did get it restarted. If it did, then the regular check from PING should keep it working.

Ping isn't bringing it back. Happened again over the weekend and the only way I could get it back was with the HA rebuild core

sofkaski, 2024-04-30:
There is a lengthy discussion in #523 where it seems that there are couple of variations of the  connection fault.
My inverter is hybrid Deye SUN-8K-SG04P3-EU with a logger with firmware version LSW3_15_FFFF_1.0.9E. I have connected it also with Deye cloud. 

After investigating the problem in my environment, I ended up proposing some changes to `pysolarmanv5` library that came out in release 3.0.2. It is likely that the version does not solve all connection problems. Particularly since I think there is also some mystery on the logger side.  The new version is anyways worth trying.

jouster1974, 2024-04-30:
Just to update , mine has probably lost connection and regained it 7 or 8 times in the past hour whilst all other entities are working fine...and the whole time, the solids cloud has got all the information correctly without dropping out.

Madridista09, 2024-05-11:
Reverting Solarman HACS integration to version 1.4.3 did the trick for me.