StephanJoubert/home_assistant_solarman #573: Solarmann daily production max 100KWh

diseco, 2024-04-30:
it is possible to increase the maximum production (today daily production ? . Currently it looks as if the max was set to 100kWh.

finn76HH, 2024-04-30:
Same here, exactly at 100kWh the integration stopped working... 

[image]

Unfortunately this makes it necessary to add the real kWh using the statistics tool in order to not loose or corrupt any long time data of the solar production.

jmccrohan, 2024-05-01:
Which inverter definition file are you using?

diseco, 2024-05-01:
deye_sg04lp3.yaml

jmccrohan, 2024-05-01:
> deye_sg04lp3.yaml

There is your issue right [there](https://github.com/StephanJoubert/home_assistant_solarman/blob/main/custom_components/solarman/inverter_definitions/deye_sg04lp3.yaml#L102). Suggest you file a PR to increase or remove the 100kWh max value.