StephanJoubert/home_assistant_solarman #574: Serial number 27*********

sandridgepark, 2024-04-30:
Hello 
I have a serial number starting Serial number 27*********
firmware LSW3_32U_5406_1.06

I cannot for the life of me get this to connect

Anyone know what file I should be using

THANK YOU !

gavrik, 2024-05-01:
The same for me. Can't connect.

gavrik, 2024-05-01:
Seems that pysolarmanv5 library is not support 27*** serial number for the solarman logger.
`
modbus = PySolarmanV5("192.168.88.206", 2796911391)
print(modbus.read_holding_registers(1476, 1))
`
And error was 
`
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 551, in read_holding_registers
    modbus_values = self._get_modbus_response(mb_request_frame)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 427, in _get_modbus_response
    mb_response_frame = self._send_receive_modbus_frame(mb_request_frame)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 415, in _send_receive_modbus_frame
    mb_response_frame = self._v5_frame_decoder(v5_response_frame)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 256, in _v5_frame_decoder
    raise V5FrameError("V5 frame does not contain a valid Modbus RTU frame")
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame does not contain a valid Modbus RTU frame
`

jmccrohan, 2024-05-01:
> Seems that pysolarmanv5 library is not support 27*** serial number for the solarman logger.

What version of pysolarmanv5 are you using? Please enable debug logging (example below) and attach the output.

```
modbus = PySolarmanV5("192.168.88.206", 2796911391, verbose=True)
```

gavrik, 2024-05-01:
I am using version 3.0.2
```
pysolarmanv5       3.0.2
```

Debug information

```python
>>> from pysolarmanv5 import PySolarmanV5
>>> modbus = PySolarmanV5("192.168.88.206", 2796911391, verbose=True)
DEBUG:pysolarmanv5.pysolarmanv5:Socket setup completed... <socket.socket fd=3, family=2, type=1, proto=6, laddr=('192.168.88.27', 60653), raddr=('192.168.88.206', 8899)>
>>> print(modbus.read_holding_registers(1476, 1))
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 ae 00 1f 7b b5 a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 05 c4 00 01 c5 3b df 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 10 00 10 15 ae 6a 1f 7b b5 a6 02 01 2f a7 25 00 ec 20 00 00 14 f9 0c 66 05 00 d0 15
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 551, in read_holding_registers
    modbus_values = self._get_modbus_response(mb_request_frame)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 427, in _get_modbus_response
    mb_response_frame = self._send_receive_modbus_frame(mb_request_frame)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 415, in _send_receive_modbus_frame
    mb_response_frame = self._v5_frame_decoder(v5_response_frame)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5.py", line 256, in _v5_frame_decoder
    raise V5FrameError("V5 frame does not contain a valid Modbus RTU frame")
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame does not contain a valid Modbus RTU frame
>>> DEBUG:pysolarmanv5.pysolarmanv5:[2796911391] COUNTER: a5 01 00 10 47 ae 6b 1f 7b b5 a6 00 66 15
```

jmccrohan, 2024-05-02:
> I am using version 3.0.2
>
> ```
> pysolarmanv5       3.0.2
> ```
>
> Debug information
>
> ```python
> DEBUG:pysolarmanv5.pysolarmanv5:Socket setup completed... <socket.socket fd=3, family=2, type=1, proto=6, [...]

Are you using a Sofar inverter by any chance? Another user reported a [similar issue](https://github.com/jmccrohan/pysolarmanv5/issues/55) earlier today but it was found that the wrong Modbus function code was being used.

gavrik, 2024-05-03:
I am using Deye 3 phase 12kwat inverter.

tragram, 2024-05-06:
Interestingly, the [deye-controller](https://github.com/githubDante/deye-controller/tree/master) library which is also based on pysolarmanv5 can read all the data correctly from my Deye SUN-10K-SG04LP3-EU even with a 27xxx logger, so the issue appears to be in this library/configuration...

gavrik, 2024-05-06:
Confirm. deye-controller can read inverter with 27xxxx serial number.

```python
>>> from deye_controller import HoldingRegisters, WritableRegisters
>>> from pysolarmanv5 import PySolarmanV5
>>> inv = PySolarmanV5('192.168.88.206', 2796911391)
>>> register = HoldingRegisters.BMSBatteryCapacity
>>> res = inv.read_holding_registers(register.address, register.len)
>>> register.value = res[0] if register.len == 1 else res
>>> print(register.description, register.format(), register.suffix)
bms_battery_SOC 100 %
>>> inv.disconnect()
```

githubDante, 2024-05-07:
@gavrik  `print(modbus.read_holding_registers(1476, 1))` - 1476 is not a valid address for Deye. What you receive in the response is an exception code 0x05 which indeed is not a valid Modbus frame.

See [here](https://github.com/StephanJoubert/home_assistant_solarman/issues/576#issuecomment-2094286126) how to enable the debug logging if you still have connectivity issues.

gavrik, 2024-05-07:
@githubDante  Thanks for you support. I recheck the integration, remove and add integration with deye_sg04lp3.yaml config and I happy can say that it is working! I can see all parameters what I need and it is changes. I have 3Phase 12kwatt inverter
Thanks.

jmccrohan, 2024-05-07:
> @githubDante Thanks for you support. I recheck the integration, remove and add integration with deye_sg04lp3.yaml config and I happy can say that it is working! I can see all parameters what I need and it is changes. I have 3Phase 12kwatt inverter Thanks.

Glad to hear. It sounds like pysolarman v3.0.2 has fixed the issue for you. Hopefully @StephanJoubert can get around to releasing a new version of the Solarman integration soon with a versioned depends on pysolarmanv5.