StephanJoubert/home_assistant_solarman #576: Sensor states Unknown after 4 days uptime

markogresak, 2024-05-03:
My correction is working correctly, but after around 4 days of uptime, all sensors go to Unknown, even though `Status_connection` is still Connected. When I restart Home Assistant, the state is correct again. 

I am using v1.5.1 with definition deye_sg04lp3.yaml.

I did not find anything useful in the logs.

Can this be fixed, or is there at least a workaround?

PV-Joe, 2024-05-03:
I recommend Solarman 1.4.3.
It runs very stable with my Deye Hybrid 8K since 1,5 years with an update interval of 20 seconds.

jmccrohan, 2024-05-04:
> My correction is working correctly, but after around 4 days of uptime, all sensors go to Unknown, even though `Status_connection` is still Connected. When I restart Home Assistant, the state is [...]

Please enable debug logging and attach same.

Add the following to `configuration.yml`:
```
logger:
  logs:
     custom_components.solarman: debug
```

markogresak, 2024-05-10:
Still running the version v1.5.1, this is the log output, the update interval was set to 15s:

<details>
<summary>Log output</summary>

```log
2024-05-10 13:11:39.533 DEBUG (SyncWorker_13) [custom_components.solarman.solarman] All queries succeeded, exposing updated values.
2024-05-10 13:12:05.354 DEBUG (SyncWorker_53) [custom_components.solarman.solarman] Starting to query for [6] ranges...
2024-05-10 13:12:05.354 DEBUG (SyncWorker_53) [custom_components.solarman.solarman] Querying [3 - 89]...
2024-05-10 13:12:05.355 DEBUG (SyncWorker_53) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 3e 00 a6 f7 80 a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 57 f4 34 f1 15
2024-05-10 13:12:05.893 DEBUG (SyncWorker_53) [custom_components.solarman.solarman] RECD: a5 c1 00 10 15 3e 8a a6 f7 80 a2 02 01 c4 e5 18 01 72 02 00 00 40 1b 25 65 01 03 ae 32 32 31 31 31 36 39 35 33 30 00 00 00 00 00 00 18 07 00 00 00 00 20 05 11 44 00 00 10 01 c0 37 00 00 d4 c0 00 01 ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff 18 05 0a 0d 0c 08 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 01 ff ff 00 01 ff ff 00 00 03 e8 00 01 00 00 ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 8a 59 62 15
2024-05-10 13:12:05.895 DEBUG (SyncWorker_53) [custom_components.solarman.solarman] Querying [3 - 89] succeeded
2024-05-10 13:12:05.896 DEBUG (SyncWorker_53) [custom_components.solarman.solarman] Querying [514 - 558]...
2024-05-10 13:12:05.896 DEBUG (SyncWorker_53) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 3f 00 a6 f7 80 a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 02 00 2d 25 af 75 15
2024-05-10 13:12:15.355 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solarman_pv1_power is taking over 10 seconds
2024-05-10 13:12:20.911 DEBUG (SyncWorker_53) [custom_components.solarman.solarman] Got exception when receiving frame
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5.py", line 277, in _send_receive_v5_frame
    v5_response = self._data_queue.get(timeout=self.socket_timeout)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/queues.py", line 114, in get
    raise Empty
_queue.Empty
2024-05-10 13:12:20.914 WARNING (SyncWorker_53) [custom_components.solarman.solarman] Querying [514 - 558] failed with exception [Empty: ]
2024-05-10 13:12:20.914 INFO (SyncWorker_53) [custom_components.solarman.solarman] Disconnecting from solarman data logger obfuscated-logger-ip:8899
2024-05-10 13:12:20.920 WARNING (SyncWorker_53) [custom_components.solarman.solarman] Querying [514 - 558] failed, [1] retry attempts left
2024-05-10 13:12:20.920 INFO (SyncWorker_53) [custom_components.solarman.solarman] Connecting to solarman data logger obfuscated-logger-ip:8899
2024-05-10 13:12:22.975 DEBUG (SyncWorker_53) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=12, family=2, type=1, proto=6, laddr=('obfuscated-HA-ip', 44372), raddr=('obfuscated-logger-ip', 8899)>
2024-05-10 13:12:22.976 DEBUG (SyncWorker_53) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 06 00 a6 f7 80 a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 02 00 2d 25 af 3c 15
2024-05-10 13:12:35.355 WARNING (MainThread) [homeassistant.components.sensor] Updating solarman sensor took longer than the scheduled update interval 0:00:30
2024-05-10 13:12:37.992 DEBUG (SyncWorker_53) [custom_components.solarman.solarman] Got exception when receiving frame
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5.py", line 277, in _send_receive_v5_frame
    v5_response = self._data_queue.get(timeout=self.socket_timeout)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/queues.py", line 114, in get
    raise Empty
_queue.Empty
2024-05-10 13:12:37.994 WARNING (SyncWorker_53) [custom_components.solarman.solarman] Querying [514 - 558] failed with exception [Empty: ]
2024-05-10 13:12:37.995 INFO (SyncWorker_53) [custom_components.solarman.solarman] Disconnecting from solarman data logger obfuscated-logger-ip:8899
2024-05-10 13:12:37.999 WARNING (SyncWorker_53) [custom_components.solarman.solarman] Querying [514 - 558] failed, [0] retry attempts left
2024-05-10 13:12:38.000 WARNING (SyncWorker_53) [custom_components.solarman.solarman] Querying registers [514 - 558] failed, aborting.
2024-05-10 13:12:38.001 DEBUG (SyncWorker_53) [custom_components.solarman.sensor] No value recorded for PV1 Power
2024-05-10 13:12:38.005 DEBUG (SyncWorker_20) [custom_components.solarman.sensor] No value recorded for PV2 Power
2024-05-10 13:12:38.009 DEBUG (SyncWorker_29) [custom_components.solarman.sensor] No value recorded for PV1 Voltage
2024-05-10 13:12:38.012 DEBUG (SyncWorker_33) [custom_components.solarman.sensor] No value recorded for PV2 Voltage
2024-05-10 13:12:38.014 DEBUG (SyncWorker_48) [custom_components.solarman.sensor] No value recorded for PV1 Current
2024-05-10 13:12:38.018 DEBUG (SyncWorker_25) [custom_components.solarman.sensor] No value recorded for PV2 Current
[... cut ...]
2024-05-10 13:13:38.738 WARNING (SyncWorker_5) [custom_components.solarman.solarman] Querying [514 - 558] failed, [1] retry attempts left
2024-05-10 13:13:38.738 INFO (SyncWorker_5) [custom_components.solarman.solarman] Connecting to solarman data logger obfuscated-logger-ip:8899
```

</details>

If relevant, the logger is using a wifi connection, but the signal is strong. 

> The update interval of 20 seconds was edited manually.

Based on this suggestion, I have changed the config to use 30s interval, I am testing now if it will help.

d-eggert, 2024-05-13:
I have the very same issue. @markogresak did changing the update interval fix it?

markogresak, 2024-05-15:
No, I am experiencing the same problems with longer interval. 

I am now testing with reinstalled version, first trying the latest (main), if that does not work, I will switch to 1.4.3.

donzeg, 2024-05-28:
> Please enable debug logging and attach same.
>
> Add the following to `configuration.yml`:
>
> ```
> logger:
>   logs:
>      custom_components.solarman: debug
> ```

[home-assistant_solarman_2024-05-28T20-38-17.572Z.log](https://github.com/StephanJoubert/home_assistant_solarman/files/15475796/home-assistant_solarman_2024-05-28T20-38-17.572Z.log)

markogresak, 2024-05-29:
Update:

None of the version I've tried helped, and the change to 30s interval did not fix it.

I've mentioned the problem to my PV system installer. He saw the brief disconnection problem with other loggers as well, even when logging to the Solarman online service with an interval of 5 minutes. This leads me to believe there's a general networking problem with the wifi logger.

I've explored the option to use Modebus, but then found the option to have a [Solarman Stick Logger](https://www.solarmanpv.com/products/data-logger/stick-logger/) with ethernet connection. Unfortunately, after installing it, I've learned it does not work with this plugin, the connection is timing out. It's a problem with pysolarmanv5 module, reported under issue https://github.com/jmccrohan/pysolarmanv5/issues/22.

I have switched to https://github.com/kellerza/sunsynk. The solarman driver (pysolarmanv5) will not work, I've configured the connection with pymodbus driver and it's working will with an uptime of over 1 week so far.

I am closing this issue as it appears to be unrelated with this integration or pysolarmanv5 module, there's a networking problem with wifi logger stick itself.

closed 2024-05-29 by markogresak