StephanJoubert/home_assistant_solarman #579: Integration stopped working for Deye. Querying [514 - 558] failed with exception

NosIreland, 2024-05-08:
Hi,
After couple months of work the integration suddenly stooped working today. 
Looking at the logs the communication is working but it seem that the data cannot be decoded? 
All entities went into Unknown state
status_connection showing as disconnected

Inverter serial: 23**
Inverter model: 12K-SG04LP3-EU
Logger firmware: LSW3_32U5406_1.06

```
2024-05-08 21:10:05.862 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Starting to query for [6] ranges...
2024-05-08 21:10:05.862 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [3 - 89]...
2024-05-08 21:10:05.862 INFO (SyncWorker_1) [custom_components.solarman.solarman] Connecting to solarman data logger 192.168.8.250:8899
2024-05-08 21:10:05.866 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=20, family=2, type=1, proto=6, laddr=('172.18.0.4', 39950), raddr=('192.168.8.250', 8899)>
2024-05-08 21:10:05.867 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 80 00 03 eb 4c a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 57 f4 34 54 15
2024-05-08 21:10:06.203 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] RECD: a5 c1 00 10 15 80 7d 03 eb 4c a6 02 01 2d 99 d6 00 08 1e 00 00 d0 26 65 65 01 03 ae 32 33 30 39 31 31 30 35 30 32 00 00 00 00 00 00 18 07 00 00 00 00 20 06 11 40 00 00 10 01 c0 3e 00 00 d4 c0 00 01 02 03 00 00 00 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff 18 05 08 15 08 14 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 01 00 64 00 64 ff ff 00 00 04 4c 00 01 00 00 ff ff ff ff ff ff 00 00 00 01 ff ff ff ff ff ff 4b cc 9a 15
2024-05-08 21:10:06.204 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [3 - 89] succeeded
2024-05-08 21:10:06.205 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [514 - 558]...
2024-05-08 21:10:06.205 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 81 00 03 eb 4c a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 02 00 2d 25 af d8 15
2024-05-08 21:10:06.487 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] RECD: a5 6d 00 10 15 81 7e 03 eb 4c a6 02 01 2d 99 d6 00 09 1e 00 00 d0 26 65 65 01 03 5a 00 49 00 2c 31 67 00 00 2d 8a 00 00 00 02 02 f4 66 e1 00 00 5f 88 00 00 00 d0 8d 43 00 00 04 10 00 00 00 00 00 00 00 00 9c 47 00 00 00 00 00 00 00 00 00 00 04 e2 05 99 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 01 00 05 00 00 00 00 00 00 00 00 00 00 00 00 83 74 61 15
2024-05-08 21:10:06.488 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [514 - 558] failed with exception [ValueError: Invalidate complete dataset (Daily Production ~ 104.0)]
2024-05-08 21:10:06.489 INFO (SyncWorker_1) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192.168.8.250:8899
2024-05-08 21:10:06.491 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [514 - 558] failed, [1] retry attempts left
2024-05-08 21:10:06.491 INFO (SyncWorker_1) [custom_components.solarman.solarman] Connecting to solarman data logger 192.168.8.250:8899
2024-05-08 21:10:06.495 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=20, family=2, type=1, proto=6, laddr=('172.18.0.4', 39962), raddr=('192.168.8.250', 8899)>
2024-05-08 21:10:06.495 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 4d 00 03 eb 4c a6 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 02 00 2d 25 af a4 15
2024-05-08 21:10:06.668 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] RECD: a5 6d 00 10 15 4d 7f 03 eb 4c a6 02 01 2d 99 d6 00 09 1e 00 00 d0 26 65 65 01 03 5a 00 49 00 2c 31 67 00 00 2d 8a 00 00 00 02 02 f4 66 e1 00 00 5f 88 00 00 00 d0 8d 43 00 00 04 10 00 00 00 00 00 00 00 00 9c 47 00 00 00 00 00 00 00 00 00 00 04 e2 05 99 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 01 00 05 00 00 00 00 00 00 00 00 00 00 00 00 83 74 2e 15
2024-05-08 21:10:06.669 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [514 - 558] failed with exception [ValueError: Invalidate complete dataset (Daily Production ~ 104.0)]
2024-05-08 21:10:06.669 INFO (SyncWorker_1) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192.168.8.250:8899
2024-05-08 21:10:06.671 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [514 - 558] failed, [0] retry attempts left
2024-05-08 21:10:06.671 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying registers [514 - 558] failed, aborting.
2024-05-08 21:10:06.671 DEBUG (SyncWorker_1) [custom_components.solarman.sensor] No value recorded for PV1 Power
2024-05-08 21:10:06.674 DEBUG (SyncWorker_62) [custom_components.solarman.sensor] No value recorded for PV2 Power
2024-05-08 21:10:06.675 DEBUG (SyncWorker_22) [custom_components.solarman.sensor] No value recorded for PV1 Voltage
2024-05-08 21:10:06.677 DEBUG (SyncWorker_28) [custom_components.solarman.sensor] No value recorded for PV2 Voltage
2024-05-08 21:10:06.678 DEBUG (SyncWorker_35) [custom_components.solarman.sensor] No value recorded for PV1 Current
2024-05-08 21:10:06.679 DEBUG (SyncWorker_35) [custom_components.solarman.sensor] No value recorded for PV2 Current
2024-05-08 21:10:06.681 DEBUG (SyncWorker_27) [custom_components.solarman.sensor] No value recorded for Daily Production
2024-05-08 21:10:06.683 DEBUG (SyncWorker_15) [custom_components.solarman.sensor] No value recorded for Total Production
2024-05-08 21:10:06.684 DEBUG (SyncWorker_7) [custom_components.solarman.sensor] No value recorded for Daily Battery Charge
2024-05-08 21:10:06.685 DEBUG (SyncWorker_2) [custom_components.solarman.sensor] No value recorded for Daily Battery Discharge
2024-05-08 21:10:06.687 DEBUG (SyncWorker_37) [custom_components.solarman.sensor] No value recorded for Total Battery Charge
2024-05-08 21:10:06.689 DEBUG (SyncWorker_8) [custom_components.solarman.sensor] No value recorded for Total Battery Discharge
2024-05-08 21:10:06.690 DEBUG (SyncWorker_40) [custom_components.solarman.sensor] No value recorded for Battery Power
2024-05-08 21:10:06.692 DEBUG (SyncWorker_42) [custom_components.solarman.sensor] No value recorded for Battery Voltage
2024-05-08 21:10:06.693 DEBUG (SyncWorker_43) [custom_components.solarman.sensor] No value recorded for Battery SOC
2024-05-08 21:10:06.694 DEBUG (SyncWorker_10) [custom_components.solarman.sensor] No value recorded for Battery Current
2024-05-08 21:10:06.696 DEBUG (SyncWorker_41) [custom_components.solarman.sensor] No value recorded for Battery Temperature
2024-05-08 21:10:06.697 DEBUG (SyncWorker_17) [custom_components.solarman.sensor] No value recorded for Total Grid Power
2024-05-08 21:10:06.699 DEBUG (SyncWorker_51) [custom_components.solarman.sensor] No value recorded for Grid Voltage L1
2024-05-08 21:10:06.700 DEBUG (SyncWorker_48) [custom_components.solarman.sensor] No value recorded for Grid Voltage L2
2024-05-08 21:10:06.702 DEBUG (SyncWorker_52) [custom_components.solarman.sensor] No value recorded for Grid Voltage L3
2024-05-08 21:10:06.703 DEBUG (SyncWorker_50) [custom_components.solarman.sensor] No value recorded for Internal CT L1 Power
2024-05-08 21:10:06.705 DEBUG (SyncWorker_50) [custom_components.solarman.sensor] No value recorded for Internal CT L2 Power
2024-05-08 21:10:06.706 DEBUG (SyncWorker_9) [custom_components.solarman.sensor] No value recorded for Internal CT L3 Power
2024-05-08 21:10:06.708 DEBUG (SyncWorker_9) [custom_components.solarman.sensor] No value recorded for External CT L1 Power
2024-05-08 21:10:06.709 DEBUG (SyncWorker_21) [custom_components.solarman.sensor] No value recorded for External CT L2 Power
2024-05-08 21:10:06.711 DEBUG (SyncWorker_5) [custom_components.solarman.sensor] No value recorded for External CT L3 Power
2024-05-08 21:10:06.712 DEBUG (SyncWorker_56) [custom_components.solarman.sensor] No value recorded for Daily Energy Bought
2024-05-08 21:10:06.714 DEBUG (SyncWorker_57) [custom_components.solarman.sensor] No value recorded for Total Energy Bought
2024-05-08 21:10:06.715 DEBUG (SyncWorker_57) [custom_components.solarman.sensor] No value recorded for Daily Energy Sold
[... cut ...]
2024-05-08 21:10:06.743 DEBUG (SyncWorker_8) [custom_components.solarman.sensor] No value recorded for AC Temperature
2024-05-08 21:10:06.745 DEBUG (SyncWorker_8) [custom_components.solarman.sensor] No value recorded for Inverter ID
2024-05-08 21:10:06.746 DEBUG (SyncWorker_16) [custom_components.solarman.sensor] No value recorded for Communication Board Version No.
2024-05-08 21:10:06.748 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for Control Board Version No.
2024-05-08 21:10:06.749 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for Alert
```

NosIreland, 2024-05-08:
Found the issue. Apparently the max Daily production cannot be greater than 100kWh. In my case it was 104kWh
` Querying [514 - 558] failed with exception [ValueError: Invalidate complete dataset (Daily Production ~ 104.0)]`
Had to change the limit in `deye_sg04lp3.yaml`

closed 2024-05-09 by NosIreland

Joka69, 2024-05-09:
> Found the issue. Apparently the max Daily production cannot be greater than 100kWh. In my case it was 104kWh ` Querying [514 - 558] failed with exception [ValueError: Invalidate complete dataset (Daily Production ~ 104.0)]` Had to change the limit in `deye_sg04lp3.yaml`

Hey there,

Im faced with the same issue now on my Solarman integration, not sure why its happening now .

2 Questions - Where can i find the logs (running HA on Virtual box) for Solarman to try and isolate the problem.

2nd - What did you change your limit to?

NosIreland, 2024-05-09:
> Hey there,
>
> Im faced with the same issue now on my Solarman integration, not sure why its happening now .
>
> 2 Questions - Where can i find the logs (running HA on Virtual box) for Solarman to try and [...]

1. There are couple ways to get logs:
First option:
Enabled debug logging on Solarman Integration
[image]
Let it run and then disable, once it is disabled you get logs downloaded

Second option:
Enabled debug logging on Solarman Integration
Then tail HA logs `tail -f /config/home-assistant.log` your path will be different depending where/how your HA.
This option lets you see logs as they happening.

2. Since different use different registers this is why definition files are used. In my case it is `deye_sg04lp3.yaml`. The files are located in `/config/custom_components/solarman/inverter_definitions` again the path depends where your HA is installed.
In the definition file there are some limits defined and if the values are out of bounds then integration assume that the values are no valid and thus shows then as `Unknown` . In my case it was
```
    - name: "Daily Production"
      class: "energy"
      state_class: "total_increasing"
      uom: "kWh"
      scale: 0.1
      rule: 1
      registers: [0x0211]
      icon: 'mdi:solar-power'
      validation:
        max: 100
        invalidate_all:
```
I've changed the max value to 150 saved changes and restarted HA. After that all started working.

Joka69, 2024-05-09:
Aah found it, thank you. This is the error im getting:

> ```
> 2024-05-09 13:20:37.844 DEBUG (MainThread) [custom_components.solarman] __init__.py:async_unload_entry({'entry_id': '24d74e25db56ffc51f735478d50b3ae4', 'version': 1, 'minor_version': 1, 'domain': 'solarman', 'title': '192.168.10.197', 'data': {'name': 'Solarman', 'inverter_host': '192.168.10.197', 'inverter_serial': 2711484655, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_hybrid.yaml'}, 'options': {'name': 'Solarman', 'inverter_host': '192.168.10.197', 'inverter_serial': 2711484655, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_hybrid.yaml'}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'unique_id': None, 'disabled_by': None})
> 2024-05-09 13:20:37.875 WARNING (MainThread) [homeassistant.components.energy.sensor] Found unexpected state_class None for sensor.solarman_daily_load_consumption
> 2024-05-09 13:20:37.905 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry 192.168.10.** for solarman
> Traceback (most recent call last):
> File "/usr/src/homeassistant/homeassistant/config_entries.py", line 772, in async_unload
> result = await component.async_unload_entry(hass, self)
> ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> File "/config/custom_components/solarman/__init__.py", line 28, in async_unload_entry
> hass.data[DOMAIN].pop(entry.entry_id)
> ~~~~~~~~~^^^^^^^^
> KeyError: 'solarman'
> ```

NosIreland, 2024-05-11:
If it fails just on `solarman_daily_load_consumption` try to comment it out and see if it can pull other ones.