StephanJoubert/home_assistant_solarman #580: Problem with LSE-3 and DEYE SUN-12K-SG04LP3-EU. Device Serial 2109822xxx

paasherbert, 2024-05-09:
i added the Solarman Integration but i can not retrieve any data from LSE-3 Logger (LAN Logger Stick).
PS: i changed the last 2 digits from the device serial number 2109822755
Please help me out. Let me know if i can help.

Here the logfile:

2024-05-09 16:49:15.586 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration solarman which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-09 16:49:15.587 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-09 16:49:20.672 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-05-09 16:49:20.707 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.helpers.event. This is deprecated and will stop working in Home Assistant 2024.11, it should be updated to import functions used from event directly at custom_components/hacs/base.py, line 630: self.hass.helpers.event.async_track_time_interval(, please create a bug report at https://github.com/hacs/integration/issues
2024-05-09 16:49:25.973 DEBUG (MainThread) [custom_components.solarman.config_flow] config_flow.py:ConfigFlow.async_step_user: None
2024-05-09 16:49:25.973 DEBUG (MainThread) [custom_components.solarman.config_flow] config_flow.py:step_user_data_schema: {'name': 'Solarman', 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_hybrid.yaml'}
2024-05-09 16:49:25.976 DEBUG (MainThread) [custom_components.solarman.config_flow] config_flow.py:step_user_data_schema: STEP_USER_DATA_SCHEMA: {'name': <class 'str'>, 'inverter_host': <class 'str'>, 'inverter_serial': <class 'int'>, 'inverter_port': <class 'int'>, 'inverter_mb_slaveid': <class 'int'>, 'lookup_file': In(['deye_2mppt.yaml', 'deye_4mppt.yaml', 'deye_hybrid.yaml', 'deye_sg04lp3.yaml', 'deye_string.yaml', 'kstar_hybrid.yaml', 'sofar_g3hyd.yaml', 'sofar_hyd3k-6k-es.yaml', 'sofar_lsw3.yaml', 'sofar_wifikit.yaml', 'solis_1p8k-5g.yaml', 'solis_3p-4g.yaml', 'solis_hybrid.yaml', 'solis_s6-gr1p.yaml', 'zcs_azzurro-ktl-v3.yaml', 'custom_parameters.yaml'])}
2024-05-09 16:49:50.214 DEBUG (MainThread) [custom_components.solarman.config_flow] config_flow.py:ConfigFlow.async_step_user: {'name': 'Solarman', 'inverter_host': '192.168.178.21', 'inverter_serial': 2109822755, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_sg04lp3.yaml'}
2024-05-09 16:49:50.215 DEBUG (MainThread) [custom_components.solarman.config_flow] config_flow.py:validate_input: {'name': 'Solarman', 'inverter_host': '192.168.178.21', 'inverter_serial': 2109822755, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_sg04lp3.yaml'}
2024-05-09 16:49:50.215 DEBUG (MainThread) [custom_components.solarman.config_flow] config_flow.py:ConfigFlow.async_step_user: validation passed: {'name': 'Solarman', 'inverter_host': '192.168.178.21', 'inverter_serial': 2109822755, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_sg04lp3.yaml'}
2024-05-09 16:49:50.217 DEBUG (MainThread) [custom_components.solarman] __init__.py:async_setup_entry({'entry_id': 'f48cbf3bfdc6cfcbf10bd096ecb7040c', 'version': 1, 'minor_version': 1, 'domain': 'solarman', 'title': '192.168.178.21', 'data': {'name': 'Solarman', 'inverter_host': '192.168.178.21', 'inverter_serial': 2109822755, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_sg04lp3.yaml'}, 'options': {'name': 'Solarman', 'inverter_host': '192.168.178.21', 'inverter_serial': 2109822755, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_sg04lp3.yaml'}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'unique_id': None, 'disabled_by': None})
2024-05-09 16:49:50.245 DEBUG (MainThread) [custom_components.solarman.sensor] sensor.py:async_setup_entry: {'name': 'Solarman', 'inverter_host': '192.168.178.21', 'inverter_serial': 2109822755, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_sg04lp3.yaml'}
2024-05-09 16:49:50.245 DEBUG (MainThread) [custom_components.solarman.sensor] sensor.py:async_setup_platform: {'name': 'Solarman', 'inverter_host': '192.168.178.21', 'inverter_serial': 2109822755, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_sg04lp3.yaml'}
2024-05-09 16:49:50.386 DEBUG (MainThread) [custom_components.solarman.sensor] sensor.py:_do_setup_platform: async_add_entities
2024-05-09 16:49:50.386 DEBUG (MainThread) [custom_components.solarman.sensor] [<entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>, <entity unknown.unknown=unknown>]
2024-05-09 16:50:20.462 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Starting to query for [6] ranges...
2024-05-09 16:50:20.462 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [3 - 89]...
2024-05-09 16:50:20.463 INFO (SyncWorker_1) [custom_components.solarman.solarman] Connecting to solarman data logger 192.168.178.21:8899
2024-05-09 16:50:20.484 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [3 - 89] failed with exception [OSError: [Errno 30] Read-only file system]
2024-05-09 16:50:20.485 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [3 - 89] failed, [1] retry attempts left
2024-05-09 16:50:20.486 INFO (SyncWorker_1) [custom_components.solarman.solarman] Connecting to solarman data logger 192.168.178.21:8899
2024-05-09 16:50:20.488 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [3 - 89] failed with exception [OSError: [Errno 30] Read-only file system]
2024-05-09 16:50:20.489 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [3 - 89] failed, [0] retry attempts left
2024-05-09 16:50:20.489 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying registers [3 - 89] failed, aborting.
2024-05-09 16:50:20.490 DEBUG (SyncWorker_1) [custom_components.solarman.sensor] No value recorded for PV1 Power
2024-05-09 16:50:20.493 DEBUG (SyncWorker_8) [custom_components.solarman.sensor] No value recorded for PV2 Power
[... cut ...]
2024-05-09 16:50:20.500 DEBUG (SyncWorker_4) [custom_components.solarman.sensor] No value recorded for PV2 Current
2024-05-09 16:50:20.503 DEBUG (SyncWorker_5) [custom_components.solarman.sensor] No value recorded for Daily Production
2024-05-09 16:50:20.505 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for Total Production
2024-05-09 16:50:20.506 DEBUG (SyncWorker_6) [custom_components.solarman.sensor] No value recorded for Daily Battery Charge
2024-05-09 16:50:20.508 DEBUG (SyncWorker_7) [custom_components.solarman.sensor] No value recorded for Daily Battery Discharge

paasherbert, 2024-05-10:
seems like that is know issue. LSW work LSE don't.
Please see here: 
https://github.com/adnovea/solarman_lse

I will test this now

paasherbert, 2024-05-10:
It works:-)
I used the deye_sg04lp3.yaml for my DEYE SUN-12K-SG04LP3-EU inverter.

Please follow the instructions here:

https://github.com/adnovea/solarman_lse/blob/main/custom_components/solarman_lse/README.md

MartinNeurol, 2024-05-27:
So it does work for you? My inverter never survives more than 24 hours without falling in to COM mode error and stops sending data completely :/