StephanJoubert/home_assistant_solarman #581: Cant get the info from inverter DEYS-SUN-4k SN 23XX ans Firmware LSW3_15_FFFF_1.0.9E

felvieira, 2024-05-12:
I try everey config from yamls but none is working

[image]

[image]

[image]

and my custom_parameters

requests:
  - start: 0x0003
    end:  0x0070

parameters:
 - group: solar
   items: 
    - name: "PV1 Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x006D]
      icon: 'mdi:solar-power'

    - name: "PV2 Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x006F]
      icon: 'mdi:solar-power'

    - name: "PV1 Current"
      class: "current"
      uom: "A"
      scale: 0.1
      rule: 1
      registers: [0x006E]
      icon: 'mdi:solar-power'

    - name: "PV2 Current"
      class: "current"
      state_class: "measurement"
      uom: "A"
      scale: 0.1
      rule: 1
      registers: [0x0070]
      icon: 'mdi:solar-power'

    - name: "Daily Production"
      class: "energy"
      state_class: "measurement"
      uom: "kWh"
      scale: 0.1
      rule: 1
      registers: [0x003C]
      icon: 'mdi:solar-power'

    - name: "Total Production"
      class: "energy"
      state_class: "total_increasing"
      uom: "kWh"
      scale: 0.1
      rule: 3
      registers: [0x003F,0x0040]
      icon: 'mdi:solar-power'

    - name: "Solar DC Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 0.1
      rule: 1
      registers: [0x0052]
      icon: 'mdi:solar-power'

 - group: Grid
   items: 
    - name: "Grid Voltage L-L"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x0049]
      icon: 'mdi:transmission-tower'

 - group: Inverter
   items: 
    - name: "Running Status"
      class: ""
      state_class: ""      
      uom: ""
      scale: 1
      rule: 1
      registers: [0x003B]
      isstr: true
      lookup: 
      -  key: 0
         value: "Stand-by"
      -  key: 1
         value: "Self-checking"
      -  key: 2
         value: "Normal"
      -  key: 3
         value: "FAULT"
      icon: 'mdi:home-lightning-bolt'

    - name: "Total Ouput AC Power"
      class: "power"
      state_class: "measurement"      
      uom: "W"
      scale: 0.1
      rule: 1
      registers: [0x0050]
      icon: 'mdi:home-lightning-bolt'

    - name: "Output AC Current"
      class: "current"
      state_class: "measurement"      
      uom: "A"
      scale: 0.1
      rule: 2
      registers: [0x004C]
      icon: 'mdi:home-lightning-bolt'

    - name: "Inverter ID"
      class: ""
      state_class: ""      
      uom: ""
      scale: 1
      rule: 5
      registers: [0x0003,0x0004,0x0005,0x0006,0x0007]
      isstr: true

fejfarj, 2024-05-16:
Hi, 
same for me.
Firmware version LSW3_15_270A_1.68

gerarfv88, 2024-05-18:
Hi,

Installing Home Assistant 11.1, Solarman 1.4.3 and pysolarmanv5 3.0.2 on other versions does not work with this type of inverter.

This work for me

[image]

felvieira, 2024-05-19:
> Hi,
>
> Installing Home Assistant 11.1, Solarman 1.4.3 and pysolarmanv5 3.0.2 on other versions does not work with this type of inverter.
>
> This work for me [...]

i dont undertand, what work for you?

I only can make work with a api directly.

fejfarj, 2024-05-20:
> Hi, same for me. Firmware version LSW3_15_270A_1.68

sorry for confusion, I have inverter type HYD 20KT-3PH, not DEYS*
however, it was not working for me as well on HA Core 2024.5.3 and it works again after updating to Core 2024.5.4 today.
I'm using configuration sofar_g3hyd.yaml

route66por, 2024-05-23:
Try do install Solarman version 1.4.3 and reboot system.

felvieira, 2024-05-23:
Solarman version 1.4.3? this repo ate 1.4.3 version? thats it?

route66por, 2024-05-23:
Yes, forget the last version, try the 1.4.3 instead, reboot any then insert all the info of the WiFi dongle of the inverter.