StephanJoubert/home_assistant_solarman #582: Solis 3P6K 4G not working

Kuchiru, 2024-05-13:
Hi there,

I have a Solis 3P6K with a WiFi stick connected to my network.
The serial number is 4049800169 and the firmware MW_08_512_0501_1.82

When i add the device with the following config:
[image]

I'm unfortunately met with the following log entry:
` Querying [4999 - 4999] failed with exception [IllegalDataAddressError: The data address received in the request is not an allowable address for the server. ] `

All entities are unknown and the connection state is disconnected.

I'm not sure how to continue from here and any help is appreciated.

Kuchiru, 2024-05-13:
@agirilovich i see you committed the yaml for this device, would you have any pointers for me?

agirilovich, 2024-05-13:
Yeah. Suddenly reading of 4999 register got breakable.
I didn't investigate it deeply. But here is hotfix [remove 4999 register request](https://github.com/StephanJoubert/home_assistant_solarman/commit/ff30b215532e5821510cb176773be82d950edd07) available.

Obviously, you are using old version of component. Just update it.

closed 2024-06-11 by Kuchiru