StephanJoubert/home_assistant_solarman #583: Missing DC Power in deye_2mppt.yaml

NikromHD, 2024-05-14:
Hallo ich hab eine Frage.

Und zwar sind in Solarman auch die DC Power Leistungen hinterlegt.
Diese würde ich ebenfalls gerne in Homeassistent angezeigt bekommen.
Bei mir läuft die deye_2mppt.yaml einwandfrei

Ich habe aus der deye_hybrid.yaml die die Werte Register felder übernommen. Jedoch bekomme ich dann nur unbekannt angezeigt.

Kann mir jemand die Registers für die Werte nennen ?

Ich würde gerne auswerten wie Ost/West sich verhalten.

-----------------------------------------------------------------------------------------------------------------------------

Hello, I have a question.

The DC power outputs are also stored in Solarman.
I would also like to have these displayed in Home Assistant.
The deye_2mppt.yaml works perfectly for me

I have transferred the values of the register fields from deye_hybrid.yaml. However, I then only get unknown.

Can someone tell me the registers for the values?

I would like to evaluate how east/west behave.

[image]
[image]

Viele Grüße
Nikrom

githubDante, 2024-05-14:
It appears  that the values are calculated from Voltage * Current for the respective string. There are no register definitions for them.

NikromHD, 2024-05-22:
Thanks for the Info.

I solved this with an Helper
I created a template with the code
`{{ (states('sensor.west_ost_pv2_voltage') | float *
    states('sensor.west_ost_pv2_current') | float)| round(2) }}`

Its Look like this 
[image]

And a second template Helper for pv1 with Same Code

closed 2024-05-22 by NikromHD