StephanJoubert/home_assistant_solarman #585: HA intergration no longer working in 2024.5

fredrikchabot, 2024-05-15:
This error originated from a custom integration.

Logger: custom_components.solarman.solarman
Source: custom_components/solarman/solarman.py:96
integration: Solarman (documentation, issues)
First occurred: 14:35:09 (77 occurrences)
Last logged: 14:44:40

Querying registers [1 - 125] failed, aborting.

NJTrance, 2024-05-15:
same issue Home Assistant Core Update 2024.5.3 kills integration.

all entities status unknown.

roll back on 2024.4.2 and works fine.

NJTrance, 2024-05-15:
Solarman just stopped working #557 

this seams to solve it.

"ha core rebuild"

just wait few minutes after HA restarts.

Oberlandix, 2024-05-19:
Hi,
same issue here! Latest Core Update yesterday (2024.5.3) has killed my integration.

All entities of solarman now: status unknown.

"Querying registers (3- 89) failed, aborting.

This is very sad because I have a lots of automations and loggings active for my PV...

Looking forward to solve the problem - I ll try to roll back now; using proxmox-snapshot.

THX

fejfarj, 2024-05-20:
Hi, it works again after updating to Core 2024.5.4 today.

Oberlandix, 2024-05-22:
Thx a lot fejfarj for this information. :-)

Please forgive my layman's question: Is it coincidence or pure luck that this now works for you or could you / someone read from the changelog for the latest version whether / what has been done to this adjusting screw?

I'm thinking about cloning my Homeassistant instance to Proxmox to be on the safe side and setting up a test system here for pre-testing, like in production, so to speak....

Many thanks!

fejfarj, 2024-05-22:
It was my intention to roll back to previous core version where it was working, but suddenly I saw new core version, so I gave it one try to see if new core will remove this issue.

norb-HB, 2024-05-23:
I have core 2024.5.4 and Solarman is still not working. 

I wonder if the developer Stephan could comment if he intends to look after the issue?

norb-HB, 2024-05-23:
I'm not too keen to fiddlin around with my HA installation. However. Could please someone explain what the "ha core rebuild" actual does and if it fixes the issue for the latest core.

gigibiacco, 2024-05-25:
I have core 2024.5.5 but Solarman integration is still non worching. 
All entities: status unknown.
"ha core rebuild" has no effect.
some help please

norb-HB, 2024-05-27:
"ha core rebuild" did actually solve it for me

Ajbadenhorst, 2024-05-28:
Can confirm "ha core rebuild" in ha CLI fixed it for me. I also needed to do a HA "Restart Home Assistant" for it to work again.

Mikusina, 2024-06-02:
same here, HA and solarman doesnt work.. keep getting disconnected

JorisHeugebaert, 2024-06-03:
Same problem here after update to 2024-5-4. ha core rebuild solved it. Thx !

Oberlandix, 2024-06-04:
Somehow I don't feel sure yet, as I read contradictory things here. It works for some, but not for others. Whether this is a coincidence or the result of an update / bugfix of the core is unclear.

We don't really have any direct feedback from one of the programmers / SME yet, do we? Unfortunately I can't code, otherwise I would of course take a look at it myself... I can also donate a cup of coffee somewhere etc...

In any case, it would be great to get a definitive fix and know what's going on. I'll probably have to clone Homeassistant in my VM and run a test system for HA updates in parallel... Thank you for reading my ‘whinge’ :-)

gs-danielnikonczuk, 2024-06-04:
Ok, a funny thing just happened. I tried this:
1. removed Solarman integration and removed from HACS.
2. `ha core rebuild` via terminal
3. Add Solarman integration repo again
4. Add Solarman services again and ensure you use the correct YAML config for your inverter model.

This time, it worked. It hasn't worked for quite a while, though. There was no HA update in between. 
What's different from other tries is that I changed the Solarman config YAML. I made sure I chose the right one for my model according to the descriptions in the YAML files.

groovinhigh04, 2024-06-04:
Thanks, @gs-danielnikonczuk! 👍 Same procedure solved it for me as well

Bodoundbenni, 2024-06-04:
> Ok, a funny thing just happened. I tried this:
> 
>     1. removed Solarman integration and removed from HACS.
> 
>     2. `ha core rebuild` via terminal
> 
>     3. Add Solarman integration repo again
> 
>     4. Add Solarman services again and ensure you use the correct YAML config for your inverter model.
> 
> 
> This time, it worked. It hasn't worked for quite a while, though. There was no HA update in between. What's different from other tries is that I changed the Solarman config YAML. I made sure I chose the right one for my model according to the descriptions in the YAML files.
> 
> HA Green Core: 2024.5.5 Supervisor: 2024.05.1 Operating System: 12.3

THX for sharing! after trying and trying I simple workaround that brings my Deye back to HA

Oberlandix, 2024-07-20:
Dear friends. This evening I tried it again and updated to:
Core 2024.7.3
Supervisor 2024.06.2
Operating System 12.4 

The solarman integration is still NOT working. I am really frustrated in the meatime and at the moment my role-back is running. 

Can you please be so nice and explaine this points to me:

1. How to do the "ha core rebuild"?
2. Won't I lose all my analyses, calculations and displays if i completely remove the solarman integration and reinstall it? I have spent many hours on this...
3. Where is responsible for this bug? The developer of the solarman integration (still in v 1.0 and no update) or the coders of homeassistant? Where can I explore for help?

I am not a professional in coding, sorry - I just want to use homeassistant. As in the past. I am feeling lonely with that problem. Maybe my HA will stay in the version of may 2024 for ever?! 

Thx vor help and reading my sad message. All the best, Oberlandix