StephanJoubert/home_assistant_solarman #589: The entity is non-numeric

ftwenzel, 2024-05-30:
The entity is non-numeric once the inverter runs out of power. 
The state then changes to “Unknown”. 
Is it possible to change the sensor.solarman_total_ac_output_power_active from Unknown to zero?

icrismariu, 2024-06-01:
Same for me!

Killerus, 2024-06-28:
I have the same error. I received information that the Solarman add-on has such problems since system version 12.x

andrejooste22, 2024-10-27:
I have the same error with my Deye Inverter. Did anyone here manage to sort out the Non-metric issue?

mcderd-coder, 2024-10-30:
> I have the same error with my Deye Inverter. Did anyone here manage to sort out the Non-numeric issue?

I reported this months ago, I have been using version 1.4.3 without issue on my main system (I rolled back), on my test system I just re-installed 1.5.1 to test and it still has this error. It seems nobody has a fix for this.