StephanJoubert/home_assistant_solarman #593: Solarman version

gio-dot, 2024-06-02:
I'm sure i'm missing something but... On Github latest version is 1.5.1, but in Homeassistant integrations i see 1.0.0. Is this normal?

[image]

[image]

Chickenbreast0, 2024-06-14:
Is this a serious question? 

[image]

PCB0815, 2024-06-21:
I have the same problem and my Solarman integration is not connecting to my logger. So please give a serious answer to a beginner.

Chickenbreast0, 2024-06-21:
> I have the same problem and my Solarman integration is not connecting to my logger. So please give a serious answer to a beginner.

Please poste your Settings. How did you set it up? Lets start from the scratch...

Do you have access to your Deye by IP adress?

PCB0815, 2024-06-21:
In this moment it works. It was a problem that my logger does not work with 1.5.1 but with 1.4.3 but I had issues to downoald 1.4.3 instead of 1.5.1, now solved, Thanks for offering help!

Chickenbreast0, 2024-06-21:
> In this moment it works. It was a problem that my logger does not work with 1.5.1 but with 1.4.3 but I had issues to downoald 1.4.3 instead of 1.5.1, now solved, Thanks for offering help!

No worries. May I know if which Version you are using now?

PCB0815, 2024-06-21:
1.4.3 with a logger SN starting with 23********

ChristophCaina, 2024-06-22:
the problem might be the version info in the manifest file...

```
{
  "domain": "solarman",
  "name": "Solarman",
  "codeowners": ["@StephanJoubert"],
  "config_flow": true,
  "dependencies": [],
  "documentation": "https://github.com/StephanJoubert/home_assistant_solarman/blob/main/README.md",
  "iot_class": "local_polling",
  "issue_tracker": "https://github.com/StephanJoubert/home_assistant_solarman/issues",
  "requirements": ["pyyaml","pysolarmanv5"],
  "version": "1.0.0"
}
```