StephanJoubert/home_assistant_solarman #594: Integration showing strange values

Reciv3, 2024-06-02:
Hi Guys,

Did you notice incorrect values reported by integration?

[image]

Thanks in advance!

Point-Xero, 2024-06-18:
I am also receiving incorrect values. @Reciv3 did yours come right in the end?

Tomson-pl, 2024-06-19:
Deye 10kw inverter the same problem. ver 1.5.1

p0ptab, 2024-06-29:
I have the same problem, Deye 15kW

Reciv3, 2024-07-02:
@Point-Xero still showing incorrect values...but I did nothing to check what's wrong

piotrkazmierczak2323, 2024-07-15:
@Reciv3 what is your logger serial number? Some of them are not supported including mine - 29xxxxxxx
Any chance for a fix?

Reciv3, 2024-08-05:
[image]

Hi @piotrkazmierczak2323, @p0ptab @Tomson-pl @Point-Xero,

I have fixed my integration. Steps:
1) you need to check your inverter model (in my case SG04LP3-EU)
2) Open main solarman directory -> inverter_definitions -> and search for definitions that fits your model
- this is typically described in first 2 rows of the file
3) Go to main HA settings -> Devices and Services -> Open Solarman Integration and press configure
4) In the last section change your file (from step 2) to new one and save.
5) Restart HA

br3d, 2024-11-28:
@Reciv3 It helped, thank you!
SUN-10k inverter needs a config deye_sg04lp3.yaml