StephanJoubert/home_assistant_solarman #595: Sofar HYD 6KTL-3PH and LSW3 Logger

groovinhigh04, 2024-06-03:
All,

I am having difficulties getting my Sofar Solar HYD 6KTL-3PH inverter with LSW3 WiFi Logger running. I have selected different configurations from the presets but getting all sorts of strange values and many unknown/ not available. 

Any help would be much appreciated!

thanks in advance and kind regards 
Johannes

squozeneu, 2024-06-06:
Having the same issue with a HYD-6K-1PH. I've never managed to get a reading from it, and I've tried an HA core rebuild which made no difference. Is there a way to check logs to see why it's failing?

gothrek, 2024-06-11:
same prob, i try diff confs but i dont receive any data, only not available or unknown

avorionero, 2024-06-13:
hi,
I add in configuration yaml and, after one minute with "entity is non-numeric" now it's working 

sensor:
  - platform: solarman
    name: Fotovoltaico
    inverter_host: my inverter IP
    inverter_port: 8899
    inverter_serial: 17xxxxxxxxxx
    inverter_mb_slaveid: 1
    scan_interval: 120
    lookup_file: sofar_hyd3k-6k-es.yaml

I think is "scan interval" the helper

LotharF, 2024-07-21:
I'm using also this file and it is working with my SofarSolar HYD 8KTL 3PH. But where is the value that show me the consumption of my home?

miralpeix, 2024-08-09:
> hi, I add in configuration yaml and, after one minute with "entity is non-numeric" now it's working
>
> sensor:
>
> * platform: solarman
>   name: Fotovoltaico
>   inverter_host: my inverter IP [...]

How can I edit this valour?

avorionero, 2024-08-09:
> How can I edit this valour?

NOTE:  despite the change, I have newly "entity is non-numeric" but statistical dayly data are ok

tomosvari, 2025-08-22:
> Had similar issues, then rollback to 1.4.3 fw ans now it is working perfectly.
> 
> Logged an issue yesterday for the same.

Can i ask you, how can you rollback the firmware? I have a good functioning system till 2 weeks ago. Then my installer updated my firmware remotely and after that my system no more give any data locally. API integration in home assistant give no sensors data. With curl can i get the data but for me is that pointless. So i have try to downgrade the firmware but it stay the newer one.