StephanJoubert/home_assistant_solarman #597: Detected blocking call to open inside the event loop by custom integration 'solarman' at custom_components/solarman/solarman.py

bert1111, 2024-06-07:
Detected blocking call to open inside the event loop by custom integration 'solarman' at custom_components/solarman/solarman.py, line 33: with open(self.path + self.lookup_file) as f: (offender: /config/custom_components/solarman/solarman.py, line 33: with open(self.path + self.lookup_file) as f:), please create a bug report at https://github.com/StephanJoubert/home_assistant_solarman/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked await self.async_setup(hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/config/custom_components/solarman/__init__.py", line 18, in async_setup_entry await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2032, in async_forward_entry_setups await asyncio.gather( File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2034, in <genexpr> create_eager_task( File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2081, in _async_forward_entry_setup await entry.async_setup(self.hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 134, in async_setup_entry return await component.async_setup_entry(entry) File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 196, in async_setup_entry return await self._platforms[key].async_setup_entry(config_entry) File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 333, in async_setup_entry return await self._async_setup_platform(async_create_setup_awaitable) File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 363, in _async_setup_platform awaitable = create_eager_task(awaitable, loop=hass.loop) File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/config/custom_components/solarman/sensor.py", line 79, in async_setup_entry _do_setup_platform(hass, entry.options, async_add_entities) File "/config/custom_components/solarman/sensor.py", line 51, in _do_setup_platform inverter = Inverter(path, inverter_sn, inverter_host, inverter_port, inverter_mb_slaveid, lookup_file) File "/config/custom_components/solarman/solarman.py", line 33, in __init__ with open(self.path + self.lookup_file) as f:

nils512, 2024-06-07:
I see the same issue.
just installed it today.
HA-Version 2024.6.0
Solarman 1.5.1

Bikervossibaer, 2024-06-11:
I see the same issue
I have installed it today

Core
2024.6.1
Supervisor
2024.06.0
Operating System
12.3
Frontend
20240605.0

Detected blocking call to open inside the event loop by custom integration 'alexa_media' at custom_components/alexa_media/sensor.py, line 450: timezone = pytz.timezone( (offender: /usr/local/lib/python3.12/site-packages/pytz/__init__.py, line 108: return open(filename, 'rb')), please create a bug report at https://github.com/alandtse/alexa_media_player/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2055, in async_forward_entry_setup return await self._async_forward_entry_setup(entry, domain, True) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2081, in _async_forward_entry_setup await entry.async_setup(self.hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 134, in async_setup_entry return await component.async_setup_entry(entry) File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 196, in async_setup_entry return await self._platforms[key].async_setup_entry(config_entry) File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 333, in async_setup_entry return await self._async_setup_platform(async_create_setup_awaitable) File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 363, in _async_setup_platform awaitable = create_eager_task(awaitable, loop=hass.loop) File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/config/custom_components/alexa_media/sensor.py", line 162, in async_setup_entry return await async_setup_platform( File "/config/custom_components/alexa_media/sensor.py", line 151, in async_setup_platform return await add_devices( File "/config/custom_components/alexa_media/helpers.py", line 52, in add_devices add_devices_callback(devices, False) File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 515, in _async_schedule_add_entities_for_entry task = self.config_entry.async_create_task( File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1134, in async_create_task task = hass.async_create_task_internal( File "/usr/src/homeassistant/homeassistant/core.py", line 828, in async_create_task_internal task = create_eager_task(target, name=name, loop=self.loop) File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 649, in async_add_entities await add_func(coros, entities, timeout) File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 600, in _async_add_entities await coro File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 914, in _async_add_entity await entity.add_to_platform_finish() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1354, in add_to_platform_finish await self.async_added_to_hass() File "/config/custom_components/alexa_media/sensor.py", line 541, in async_added_to_hass self._process_raw_notifications() File "/config/custom_components/alexa_media/sensor.py", line 382, in _process_raw_notifications list(map(self._fix_alarm_date_time, self._n_dict.items())) File "/config/custom_components/alexa_media/sensor.py", line 450, in _fix_alarm_date_time timezone = pytz.timezone(
Detected blocking call to open inside the event loop by custom integration 'watchman' at custom_components/watchman/utils.py, line 258: for i, line in enumerate(open(yaml_file, encoding="utf-8")): (offender: /config/custom_components/watchman/utils.py, line 258: for i, line in enumerate(open(yaml_file, encoding="utf-8")):), please create a bug report at https://github.com/dummylabs/thewatchman/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_event [... cut ...]  37, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/config/custom_components/watchman/__init__.py", line 274, in async_on_home_assistant_started parse_config(hass, reason="HA restart") File "/config/custom_components/watchman/__init__.py", line 351, in parse_config entity_list, service_list, files_parsed, files_ignored = parse( File "/config/custom_components/watchman/utils.py", line 258, in parse for i, line in enumerate(open(yaml_file, encoding="utf-8")):

Skilsoe, 2024-06-12:
Have installed today and see same issue

coserotondo, 2024-06-21:
Same for me

raysmith844, 2024-06-23:
Same for me as well

mrboogiee, 2024-06-24:
following, experiencing the same issue

essera, 2024-06-28:
Hi - same issue to me. i wrok with dwain Dashboard more than 1 year. The issue will start for more than 4 month ago.

Laggger164, 2024-07-01:
Same for me on a fresh install of HACS and Solarman. This is the first time for me trying to get it working.
Does it still work for you guys or is this why the values I get make no sense?

bert1111, 2024-07-01:
It should still work. Mine does. The error is there for longer but is only now detected in the logs.

Skilsoe, 2024-07-01:
I tryed different sofar inverter devinitions and found one which was working. After changinig the Definition you must wait some minutes before the values are refreshed.

Laggger164, 2024-07-03:
Indeed I tried it again with the deye_sg04lp3.yaml and it works great!
Not sure what the blocking call is about then, maybe that's why it won't see all the sensors.

AndyNew2, 2024-07-17:
I see similar output but just as a warning. All values are there, no issues so far.

AndyNew2, 2024-07-17:
2024-07-17 21:32:23.980 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open with args ('/config/custom_components/solarman/inverter_definitions/zcs_azzurro_hyd3ph_6K_20K.yaml',) inside the event loop by custom integration 'solarman' at custom_components/solarman/solarman.py, line 31: with open(self.path + self.lookup_file) as f: (offender: /config/custom_components/solarman/solarman.py, line 31: with open(self.path + self.lookup_file) as f:), please create a bug report at https://github.com/StephanJoubert/home_assistant_solarman/issues
For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 318, in async_setup_platform
    await self._platforms[key].async_setup(platform_config, discovery_info)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 301, in async_setup
    await self._async_setup_platform(async_create_setup_awaitable)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 361, in _async_setup_platform
    awaitable = create_eager_task(awaitable, loop=hass.loop)
  File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task
    return Task(coro, loop=loop, name=name, eager_start=True)
  File "/config/custom_components/solarman/sensor.py", line 84, in async_setup_platform
    _do_setup_platform(hass, config, async_add_entities)
  File "/config/custom_components/solarman/sensor.py", line 54, in _do_setup_platform
    inverter = Inverter(path, inverter_sn, inverter_host, inverter_port, inverter_mb_slaveid, lookup_file)
  File "/config/custom_components/solarman/solarman.py", line 31, in __init__
    with open(self.path + self.lookup_file) as f:

amaciuc, 2024-09-10:
Any news about it ?

denistn, 2024-10-05:
Same issue here. Any news?

bert1111, 2024-10-12:
Just change to https://github.com/davidrapan/ha-solarman. This works and is maintained.

closed 2024-10-12 by bert1111