StephanJoubert/home_assistant_solarman #598: Deye Microinverter Zero feed-in (Nulleinspeisung)

cliely, 2024-06-08:
Hi everyone,

For those guys who are looking for a working solution to give the Deye microinverter e.g. 600W or 800W a zero feed-in, I would like to share my script here:

1. make a template "Nulleinspeisung" (zero feed-in):

{% set your max Power of your Inverter in Watt: %}
{% set maxInverterPower = 800 %}
{% set your max Inverter Regulation e.g. 100 or max 120 in percent: %}
{% set maxInverterRegulation = 100 %}
{If your Inverter is making an offset, you can change it here in Watt: %}
{% set inverterPowerOffset = 0 %}
{% Set here your min amount of your base load in your house in Watt: %}
{% set minSolarLimit = 100 %}
{% Here can you change the Zero Point goal e.g. -5, o or +10 in Watt: %}
{% set zeroPointGridAdj = -5 %}

{% set newSolarLimit = 0 %}
{% set newRegulationLimit = 0 %}

{% Put in your current Grid sensor in Watt: %}
{% set curGridUsage = states('sensor.tasmota_stromzahler_strom_power_curr') | float * -1 | replace('W','') | int %}
{%Put in your current Solarpower sensor in Watt e.g. a shelly 2pm (the inverter sensor is too lazy): %}
{% set curSolarProd = states('sensor.solaranlage_solar_leistung_power') | float (0) | replace('W','') | int %}

{% set newSolarLimit = (curSolarProd - (curGridUsage + zeroPointGridAdj)) %}

{% if newSolarLimit <= minSolarLimit %}
  {% set newSolarLimit = minSolarLimit %}
    {% elif newSolarLimit >= maxInverterPower %}
       {% set newSolarLimit = maxInverterPower %}
{% endif %}

{% set newRegulationLimit = ((newSolarLimit * maxInverterRegulation) / (maxInverterPower - inverterPowerOffset)) %}

{{ newRegulationLimit | round (0) }}

2. Make an automation "Nulleinspeisung" (zero feed-in):

alias: Solar Nulleinspeisung
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.solaranlage_solar_leistung_power
    to: null
    from: null
    enabled: true
condition:
  - condition: state
    entity_id: sensor.solarman_on_off_enable
    state: "ON"
action:
  - service: solarman.write_multiple_holding_registers
    metadata: {}
    data:
      register: 40
      values: "{{ states('sensor.kalkulation_nulleinspeisung') | replace('%','') }}"
  - delay:
      hours: 0
      minutes: 0
      seconds: 10
      milliseconds: 0
mode: single

3. Enjoy to seeing the Gauge Display between -5 & 5 Watts 😁👍

Bye,

Mathias

closed 2024-06-08 by cliely

sdk69, 2025-04-01:
Hi @cliely, I'm very interested in the ability to limit injection when price is negative in my country, but I don't understand how to change the "active power regulation" parameter. It looks like your solution could work if I tweak it a little bit to enable it only when prices are low/negative. However I don't understand how to set it up : can you please be more specific about how/where to create the template and the automation ? and where to find the initial values used in the script ? I have a Deye SUN-M200G4-EU-Q0 2000W and hope it would be compatible. Thank you so much !

cliely, 2025-04-01:
Hi @sdk69,

You simply need to add another condition to the "Solar Nulleinspeisung" automation. 

Condition: 

Sensor for "Energy Costs in USD for Import" > "Energy earning in USD for feed-in" 

Here is a customized code. You need to replace the two sensors with your own, which provide your current costs and income in € or USD, etc.:

`alias: Solar Nulleinspeisung
description: ""
mode: single
triggers:
  - entity_id:
      - sensor.solaranlage_solar_leistung_power
    to: null
    from: null
    enabled: true
    trigger: state
conditions:
  - condition: state
    entity_id: sensor.solarman_on_off_enable
    state: "ON"
  - condition: numeric_state
    entity_id: sensor.energy_out_price
    below: sensor.ernergy_in_price
actions:
  - metadata: {}
    data:
      register: 40
      values: "{{ states('sensor.kalkulation_nulleinspeisung') | replace('%','') }}"
    action: solarman.write_multiple_holding_registers
  - delay:
      hours: 0
      minutes: 0
      seconds: 10
      milliseconds: 0
`

Go to Devices & Services -> Helpers -> Create Helper -> Template -> Create Template for a Sensor.

Name: Zero Feed Calculation

State Template: (copy & paste "Kalkulation Nulleinspeisung")

If you don't have any sensors/entities for your prices and would like to enter them yourself, you can also go to "Numerical value input" under "create helpers" and create two.

With kind regards 

Mathias

sdk69, 2025-04-01:
Thanks for your quick answer ! 

I'm sorry but I can't make it work, this is very high level compared to my usual HA needs ^^"

I wonder if that can work with my setup without some heavy modifications because I have 3 Deye G4 2000W inverters and some variables in the code seem to be in german and I have everything in english.

I tried to create the template in parameters / devices & services / helpers and pasted your code in a captor model. 
=> it mentions sensor.tasmota_stromzahler_strom_power_curr I don't understand what it is and what could be my equivalent

I tried to create the automation in parameters / automations & scenes 
=> I have an "error in describing condition: e is undefined"

Being able to limit injection from HA would be a huge win, automation based on price would be step 2.
A lot of people are interested in this as Deye is very popular in France and we have a lot of days with negative prices

Thanks again
Romain