StephanJoubert/home_assistant_solarman #601: Solarman Failed to unload

bend94, 2024-06-14:
Hi

I have a solarman account where i can follow my energy production.
Now i want to retreive that in homeassistant
I installed via HACS solarman integration and registered my Deye SUN-M100G4 inverter.
I put the required info : IP, Serial (coming from Solarman cloud), profile etc

I can't get it work in HA , always the entities are "unavailable"

Please help me
Thx

Ballestikk, 2024-06-14:
I have the same problem, with Deye SUN 15K inverter

bend94, 2024-06-14:
ha! 
now working
don't know why !
here my config:

[image]

May be because, i went to solarman cloud , i clicked on "function" in top left
then "play" button , then "start" 

You should do it on your side , maybe it helps
regards

Ballestikk, 2024-06-15:
My set-up is the same (except SN. of LSW3 logger).
[image]

I only get "Unknown" on all entities:

bend94, 2024-06-15:
hi
click here first

[image]

then click "display"

[image]

then click "play" button

[image]

regards

Ballestikk, 2024-06-15:
Tried the "play button" now that lunches a dashboard from solarman. Still none respone in HA, all values is unkown.

bend94, 2024-06-16:
hi,
i was not sure that creating a graph will help !
sorry

smitte2703, 2024-11-05:
My solarman also don't want to load I made sure to have to right ip address but still nothing