StephanJoubert/home_assistant_solarman #602: Access to "Solar Sell" in Deye inverter

Ralle, 2024-06-15:
Hello,

I have the 12 kw Deye Hybrid inverter with SolarMan datalogger. I want to be able to set the "Solar Sell" value from HA automations. I already installed the Nordpool add-on so I can watch for energy prices and then I want to toggle "Solar Sell" accordingly.

I have searched the issues and found people talking about Modbus and the datalogger and I am not sure what I actually need to accomplish this. Many things feel a little bit implied from the conversations I've seen. I cannot understand it at least.

I have this project installed in HACS and can see values from my inverter. Works perfect. What's next?

RPJacobs, 2024-06-16:
Hello,

To achieve what you want, you can add a shell command to your YAML configuration and create a switch to toggle the "Solar Sell" mode. Here's how you can do it:

First, add these shell commands to your YAML file:

```
shell_command:
sell_on: 'python /config/python_scripts/test.py on'
sell_off: 'python /config/python_scripts/test.py off'
```

Then, add a switch configuration:

```
switch:

platform: template
switches:
sell:
unique_id: "switch.sell"
turn_on:
service: shell_command.sell_on
turn_off:
service: shell_command.sell_off
```
Next, use the following Python script in the /config/python_scripts/test.py file:

```
from pysolarmanv5 import PySolarmanV5
import sys

mode = 2
if len(sys.argv) > 1 and sys.argv[1] == 'on':
mode = 0

modbus = PySolarmanV5('x.x.x.x', sernum, port=8899, mb_slave_id=1, verbose=0)

modbus.write_multiple_holding_registers(register_addr=142, values=[mode])
```

Replace x.x.x.x with the IP address of your inverter and sernum with the serial number.

Hope this helps!

Ralle, 2024-06-16:
Thank you so much for this thorough description.

Can you elaborate on what “my yaml file” is? When setting up this project you have the option to do manual configuration. I didn’t do this. Is it a requirement? Otherwise I don’t have a yaml file do I?

RPJacobs, 2024-06-16:
https://www.home-assistant.io/docs/configuration/

Matheika, 2024-07-18:
Hello,
I followed also your instructions and it was working fine until recently.
First I noticed that my values from the BMS integration were not available anymore and then also the test.py script didn't work anymore.
Actually I have not changed anything and also didn't installed anything new.

Could something with my Modbus configuration be wrong by some other application? I read that someone had problems by using an old integration that doesn't support the newer versions of the Modbus anymore. I tried to deactivate all my Integrations but that also didn't help.

MrGreenGuyDK, 2024-07-31:
could you guys please clarify what is needed for this to work ?

I did do the config above but with any luck,

im using homeassistant with solarman integration and inverter is connected with the wifi dongle and nothing else.

I just want to toggle solar sell on and off from my HA dashboard,

Thanks.