StephanJoubert/home_assistant_solarman #604: Update only every 5min

Chickenbreast0, 2024-06-20:
Hello,

unfortunately the plugin (V.1.5.1) only receives new data from the local logger every 5 minutes.

I have already had the support team change the interval to 1 minute and the current data is pushed to the Solarman database (web) every minute, but is only displayed in Homeassistant every 5 minutes.

Does anyone have any advice?

The device is an Deye M100G4 with the latest FW 1.62. Previously with FW 1.59 it was the same outcome.

[image]

[image]

Chickenbreast0, 2024-06-20:
Further - I have access to the same G3 Model without the Relais. 

[image]

[image]

tobiashinz, 2024-06-23:
Same here with Deye SUN600G3-EU-230. It used to update more frequently.

Chickenbreast0, 2024-06-24:
The Solarman Team updated now the Data Upload Period to 1min and I can see this Update rate now in my Solarman Account.
But still it pulls the updates only every 5min to my HA. Is there any info yet, how to fix that?

sofkaski, 2024-06-24:
Are you perhaps looking at the data that comes from [short term statistics](https://data.home-assistant.io/docs/statistics/#database-tables)? There the resolution is 5 min (no matter what is collection interval).

Chickenbreast0, 2024-06-25:
> Are you perhaps looking at the data that comes from [short term statistics](https://data.home-assistant.io/docs/statistics/#database-tables)? There the resolution is 5 min (no matter what is collection interval).

No, I'm talking about the recording of the databases, but about the interval at which the data is pushed from Solarman to HA. This only happens every 5 minutes. 

Despite the fact that Deye's update rate has been changed from 5 minutes to 1 minute, the data is only displayed in Homeassistant every 5 minutes.
The Solarman web interface is now updated every minute. This is a pity.

PV-Joe, 2024-06-25:
My Deye Hybrid updates production and some other entities every 30 seconds.

I edited this line:
/homeassistant/custom_components/solarman/const.py

MIN_TIME_BETWEEN_UPDATES = timedelta(seconds=30)

Chickenbreast0, 2024-07-01:
I found the solution:

- Downgrade from 1.5.1 to 1.4.3. and reastart.
- Add the deye_2mppt.yaml config into the custom_parameters.yaml and restart.
Now you´ll get an update intervall of 60sec.

An increase of the min update Intervall does not work in my case. But that is okay. From 5min down to 1min is a huge gain for me ;)

sofkaski, 2024-07-02:
My understanding has been that `MIN_TIME_BETWEEN_UPDATES` is only used to throttle `update` method calls of `Inverter` class to be called at most with that interval.  The home assistant default sensor scan interval is 30s and it is defined [here](https://github.com/home-assistant/core/blob/b315b566e554fd8c233664217afe0d1116730fdd/homeassistant/components/sensor/__init__.py#L98).

The type definition `final` hints that it should not be overridden.  But I have defined in [a branch in my fork](https://github.com/sofkaski/home_assistant_solarman/tree/add-more-error-handling) `SCAN_INTERVAL = timedelta(seconds=600)` in `custom_components/solarman/const.py` and it sets the scan interval to 10 min. (which suits my use case best with a reasonable load and data amount) .

PV-Joe, 2024-07-02:
Just moved to this integration:
https://github.com/davidrapan/ha-solarman

The update interval is 5 seconds and works well.

billyheula, 2025-06-15:
@PV-Joe  where can i change the update intervall? Sorry for rehashing this old topic

I have installed the latest version of “solarman stick logger” and do not have the option. It's frustrating after I switched from ap to deye. (update interval 300 sec.) 

Deye SUN-M200G4-EU-Q0