StephanJoubert/home_assistant_solarman #605: SUN-10K-SG01HP3-EU-AM2 no values

bsafh, 2024-06-22:
Inverter SUN-10K-SG01HP3-EU-AM2 
Config deye_hybrid.yaml

The config flow says "success".

But all entities show "unknown".

Seems like the config is different for hybrid high voltage inverters?

The sunsyk addon has the same problem with their stable version,  only the dev version provides  deye high voltage definitions.

hacek, 2024-07-01:
Try this 
https://github.com/StephanJoubert/home_assistant_solarman/files/15198330/SUN-10K-SG01HP3-EU-am2.zip

Works for me on the same inverter

bsafh, 2024-07-02:
ok ... but how?

I unzipped the file and copied it to ../custom_components/solarman/inverter_definitions/
restartet HomeAssistant
--> the file does not show up in the configuration dialog for the inverter

just to be sure, I renamed the file to all lower case, but (as expected) no luck either.

So, how to use it?

furthermore:
I had a look at that file (as it is quite a bit smaller than deye_hybrid.yaml) and sure enough, there are quite a few things missing, namely the "Time of Use" data, inverter connecting status and others. And it has definitions for only one battery, but the inverter can use two batteries.

hacek, 2024-07-02:
change name to custom_parameters.yaml and it should work, that inverter file is based on deye-sg04lp03 lp and hp versions are a bit different

bsafh, 2024-07-03:
The full register list is this 
[MODBUSRTU三相高压储能通信规约V104（高压）(3)(1)1111_SG01-HP3-AM2.pdf](https://github.com/user-attachments/files/16090095/MODBUSRTU.V104.3.1.1111_SG01-HP3-AM2.pdf)

hacek, 2024-07-04:
Is anything working for You ??? 
I got that inverter and I modified this file from LP version ... yeah "Time of use is unknown" I don't need it so I didn't try to write it in file .... I got only 1 battery so 1 battery is working.
If You don't have anything remove all and add from the begining

bsafh, 2024-07-04:
Yes, connection works now, thanks!

jendapl, 2024-07-14:
[deye_hybrid_inverter .zip](https://github.com/user-attachments/files/16227887/deye_hybrid_inverter.zip)
This definition is tested with this invertor SUN-10K-SG04LP3, most of the data fields work as expected. Install by overwriting the same file in the \config\custom_components\solarman\inverter_definitions folder, works also after the solarman is installed. Then restart your homeassistent and you are good to go.

alarmatwork, 2024-07-16:
Hello, how did you get this file. I do have a Deye SUN-15K-SG01HP3-EU-AM2 and trying to figure out how to make it work

tomsi62, 2024-10-16:
> ok ... but how?
>
> I unzipped the file and copied it to ../custom_components/solarman/inverter_definitions/ restartet HomeAssistant --> the file does not show up in the configuration dialog for the [...]

Hi together,
where can I find that ..../custom-components/....directory???