StephanJoubert/home_assistant_solarman #609: Deye Hybrid Inverter SUN-16K-SG01LP1-EU with LSW-3 stick logger, SN starts with 239

Wikusvds, 2024-06-25:
We are trying to integrate Solarman with our HomeAssistant. It does not seem to be working at all. All components return either as unknown or unavailable (I have tried every available YAML file.)

Readme from Stephan Joubert says I need to use V4 protocol however there are no clear ways to find it. The link from an older post no longer exists. 

Please assist.

bsafh, 2024-07-03:
Same for SUN-10K-SG01HP3-EU-AM2

Serial number starts with 23xxx, no idea how to change the protocol

nacree, 2024-07-05:
I have the same issue here, was trying to install device with SN starting with 236. Also tried several YAML files, but no success. I tried to install 1.4.3 version as thought it would be for protocol version V4, but at least it does not work. Any hints?

ssimoe50, 2024-07-06:
Hi @nacree 

What version are you using? Can you shared the version and the config you set? Of course hide your particular and private data 😁

thanks for sharing please

nacree, 2024-07-07:
I was using 1.4.3 version with this yaml

https://gist.github.com/gushcs/223ca2c61af2d345238d286746617650

Now tried to update to 1.5.1 because it gives weird spikes during the day to production values... but now it got broken completely.. so let's try it again.

nacree, 2024-07-07:
There seems to be also some flaw that you cannot change the used definition file, but instead create a new instance and select correct yaml there. And better to restart home assistant often enough.

Now it seems to work again for me when reverted back to 1.4.3.