StephanJoubert/home_assistant_solarman #612: LSW5BLE no data

TekuSP, 2024-06-28:
Can't load any data... Logger info:
[image]

I am total newbie in this, I would appreciate any help.
```
2024-06-28 14:03:55.950 INFO (SyncWorker_12) [custom_components.solarman.solarman] Connecting to solarman data logger 192.168.1.14:8899
2024-06-28 14:03:55.954 DEBUG (SyncWorker_12) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=12, family=2, type=1, proto=6, laddr=('192.168.1.15', 59988), raddr=('192.168.1.14', 8899)>
2024-06-28 14:03:55.955 DEBUG (SyncWorker_12) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 81 00 9b 59 bd d0 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 01 f4 00 3d 71 d5 ed 15
2024-06-28 14:03:57.976 DEBUG (SyncWorker_12) [custom_components.solarman.solarman] RECD: a5 10 00 10 15 81 10 9b 59 bd d0 02 01 06 71 01 00 28 1e 00 00 a7 35 7d 66 05 00 cc 15
2024-06-28 14:03:57.976 WARNING (SyncWorker_12) [custom_components.solarman.solarman] Querying [500 - 560] failed with exception [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
2024-06-28 14:03:57.976 INFO (SyncWorker_12) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192.168.1.14:8899
2024-06-28 14:03:57.977 WARNING (SyncWorker_12) [custom_components.solarman.solarman] Querying [500 - 560] failed, [0] retry attempts left
2024-06-28 14:03:57.977 WARNING (SyncWorker_12) [custom_components.solarman.solarman] Querying registers [500 - 560] failed, aborting.
```

TekuSP, 2024-06-28:
[home-assistant_solarman_2024-06-28T12-06-03.115Z.log](https://github.com/user-attachments/files/16029665/home-assistant_solarman_2024-06-28T12-06-03.115Z.log)

sirhaplo, 2024-06-29:
Me too.  
With LSW5 reading registers it's not working. There is a thread on another repository where @psunny28 it receving data with a local server ( https://github.com/XtheOne/Inverter-Data-Logger/issues/47 ).

I'm try to develop a solution based on that approach

TekuSP, 2024-06-29:
@sirhaplo Let me know when you have some solution. In meanwhile, if I would replace logger with other one, what logger works with setup I own and works with this library?