StephanJoubert/home_assistant_solarman #614: Contribute to other form of input

sirhaplo, 2024-06-29:
Hello @StephanJoubert,
i have problems reading registers with LSW5 stick, but i found that i can sniff the data to the cloud platform and proxies them.
I'm reverse enginering the data that i receive. Ex:
`a5da021042292a244de5d1010b84c92c250041050000db4959660100c10e000030303030303030303030303000000000000000000000000000000000000000003c000205000000009a02000000000000da000000680000000000000000000000000000000000000018061c0a2635050200fc090c003801000021090a0029091200891364003b010000e4000000ae014e020000000000000200000020000000b2580000961700003b010000000000006a0000000000000000000000e5000000000000000000000030006400d21415004a0156322e30342e313656322e30352e313456352e313620202056302e33302e373756312e30372e303856312e30342e313556312e30342e30364b3600000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000926000a092e0012091d001209fc000c013b013800000000041200000000000008840000e91900000142000052890003ebe400000142000076fb0003ebe400000142000076fb000307340000030b000000f300005ba200024ad500007161000079ec00024b2d0000744b000079ec00024b2d0000744b00000002021400150215001400000000000001e0014a023803b603b6006a0cf80cf100000000000000000000091d001200150000000000000934093400103142000000000000000000000000000000000000000c000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001387000000000000000000000000314202040c054c15`

`{"Start": 165, "Length": 55810, "Control Code": 4162, "Serial": 10538, "Logger Serial": 609084881, "Frame Type": 1, "Status": 11, "Total Working Time": 2227776549, "Power On Time": 4261120, "Offset Time": 14371161, "header_unk_1": 6684928, "Progressive": 193, "header_unk_2": 14, "inv_unk_1": 0, "Serial Number": "M11024300152", "data_1a": "0x00000000000000000000000000000000000000003c", "data_1b": "0x000205000000009a", "data_2a": "0x02000000000000da", "data_2b": 0, "data_2c": 104, "data_3a": 0, "data_3b": "0x00000000000000000000000000000000000018061c", "hour": 10, "data_5": 9781, "data_6": 328192, "data_7a": 252, "data_7b": 9, "data_7c": 12, "data_8": 56, "data_9": 16777249, "data_10": 2314, "data_11": 41, "data_12a": 2322, "data_12b": 137, "data_13a": 4964, "data_13b": 59, "data_14a": 1, "data_14b": 228, "data_15a": 0, "data_15b": 174, "data_16": 334, "data_17a": 2, "data_17b": "0x00000000000002", "data_17c": 32, "data_17d": 178, "data_18a": 88, "data_18b": 150, "data_19a": 23, "data_19b": 59, "data_20a": 1, "data_20b": "0x0000000000006a", "data_21a": 0, "data_21b": 0, "data_21c": 0, "data_21d": 0, "data_21e": 0, "data_22a": 0, "data_22b": 229, "data_23a": 0, "data_23b": "0x00000000000000000000", "SOC": 48, "data_25": 6553810, "data_26": 5141, "data_27a": 0, "data_27b": 74, "data_28": 1, "LCD Version": "V2.04.16", "CPU1 Version": "V2.05.14", "v3": "V5.16   ", "Protocol No": "V0.30.77", "v5": "V1.07.08", "ARM Version": "V1.04.15", "DSP Version": "V1.04.06", "data_29": 19254, "data_zeros_1": "0x0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000", "Power grid voltage": 2342, "Power grid current": 10, "Consumption voltage": 2350, "Consumption current": 18, "? Voltage": 2333, "? Current": 18, "PV1 Voltage": 2556, "PV1 Current": 12, "PV1 Power": 315, "PV2 Voltage": 312, "PV2 Current": 0, "PV2 Power": 0, "Electricity to load on the day": "0x0412000000000000", "data_40": 142868480, "data_41": 3910729728, "data_42": 21102592, "data_43": 1384710147, "data_44": 3957587968, "data_45": 21102592, "data_46": 1996161027, "data_47": 3957587968, "data_48": 21102592, "data_49": 1996161027, "data_50a": 120848384, "data_50b": 51052544, "data_51": 15925248, "data_52": 1537343490, "data_53": 1255473152, "data_54": 1902182400, "data_55": 2045509634, "data_56": 1261240320, "data_57": 1951072256, "data_58": 2045509634, "data_59a": 19245, "data_59b": 0, "data_59c": 29771, "data_59d": 0, "data_59e": 2, "data_60": 532, "data_61": 21, "data_62": 533, "data_63": "0x0014000000000000", "data_64": 480, "data_65a": 330, "data_65b": 568, "data_66": 950, "data_67a": 950, "data_67b": 106, "data_67c": 3320, "AC Voltage": "0x0cf100000000000000000000", "data_69": 2333, "data_70": 18, "data_71": "0x0015000000000000", "data_72": 2356, "data_73": 154402832, "data_74": "0x3142000000000000000000000000000000000000", "data_75a": 786432, "data_75b": "0x000000000000", "data_zeros_2": "0x000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000", "AC Frequency": 0, "data_76a": "0x1387000000000000000000000000", "data_76b": 826409476, "Checksum": 12, "End Code": 5}`

I would to contribute to this project with a different source of data.
It's my first integration so i need some advices.
I was thinking to a different type of inverter class that instead of asking to the logger stick will ask to the sniffing server the last line received.
Then with the same structure of the other inverters it will use the string with registers from 0 to xxx.

Do you think it's doable or there will be better ways to integrate LSW5 sticks ?

sirhaplo, 2024-07-04:
Yes they are, but i couldn't receive data reading registers from the logger.
Until 2 days ago, when I did a massive brute force on all register and finally got some answers.

So the LSW5 stick is not a problem, the problem were my registers that are on very different ranges.

My inverter is a Renon Power IFL06, a clone of Megarevo R06LNA.
I started a new template for that inverter and i've got some results :
https://github.com/sirhaplo/home_assistant_solarman/blob/main/custom_components/solarman/inverter_definitions/renon_hybrid_iflXX.yaml

[image]

closed 2024-07-04 by sirhaplo

jabezz, 2024-12-03:
> Yes they are, but i couldn't receive data reading registers from the logger. Until 2 days ago, when I did a massive brute force on all register and finally got some answers.
>
> So the LSW5 stick is [...]

Did anyone ever get more answers or documentation than above?

I'm using a Megarevo Hybrid 8kw inverter. And these Renon registers seems to fit the bill perfectly.

Your current version of the script only caters for the L1 / PV1 registers. I'm looking to expand to get the L2 and PV2 registers in to complete the picture.
Hoping to do so without too much brute force.

sirhaplo, 2024-12-03:
It's funny because i have the technician that is mounting right now the second string of panels, so i will need PV2 too.
I will bump an update of my definition very soon, with some fixes i found meanwhile.

sirhaplo, 2024-12-03:
Pull request created, waiting for merge : https://github.com/StephanJoubert/home_assistant_solarman/pull/689