StephanJoubert/home_assistant_solarman #616: [New] Integration overhaul

davidrapan, 2024-07-01:
Hello,
I had a lot of issues with your current implementation (mainly with stability of the connection, etc.). So I was tweaking, tweaking and end up with hefty overhaul which works flawlessly (with my Deye SG04LP3, 272******)...

So I decided to share it, ofc... [ha-solarman](https://github.com/davidrapan/ha-solarman)

_There is really A LOT of changes so... I will be happy to answer any questions._

Have fun.

PV-Joe, 2024-07-02:
Hi David,

just moved from this integration to yours and can‘t believe how much better it works.

Especially the time between updates is much faster with approx. 5 seconds.

There are many new sensors available as well!
I recommend to switch to your intigration.

Many thanks and keep up the good work!

davidrapan, 2024-07-02:
That's wonderful to hear!

If you don't mind I have few questions for you:
- Do you have same device as me? Cause that will the one with majority of changes. (new sensors, etc)
- How was the process of switching? Were you "reconfiguring" or just swapped the code?

Anyway update interval is 5 - 25 seconds while it's 5 s most of the time and 25 s when Inverter sending data to the Solarman servers.

PS: I just found small bug which I probably introduced with some of the latest changes and it affects the connection, meaning it disconnects once an hour for around 10 seconds. I'll look into it as soon as I can. ;-) And I also have many more planned so stay tunned. ;-)

Have fun!

davidrapan, 2024-07-16:
Hi @jouster1974! Thank you for your kind words. And I'm glad it works nicely for you!

If you want you can rather open discussion at [Discussions](https://github.com/davidrapan/ha-solarman/discussions) and we can together polish some of your sensors to be more descriptive if you want (like for example status sensor etc.)

Anyway, thanks again and as always have fun!