StephanJoubert/home_assistant_solarman #617: Bug: Detected blocking call to open inside the event loop

philmassie, 2024-07-02:
Hello
Thanks for this great integration, its awesome.
Noticed this today. Apologies if it's logged already, or if its an error on my side.

```
2024-07-02 13:52:43.614 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open inside the event loop by custom integration 'solarman' at custom_components/solarman/solarman.py, line 31: with open(self.path + self.lookup_file) as f: (offender: /config/custom_components/solarman/solarman.py, line 31: with open(self.path + self.lookup_file) as f:), please create a bug report at https://github.com/StephanJoubert/home_assistant_solarman/issues
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked
    await self.async_setup(hass, integration=integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solarman/__init__.py", line 18, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2032, in async_forward_entry_setups
    await asyncio.gather(
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2034, in <genexpr>
    create_eager_task(
  File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task
    return Task(coro, loop=loop, name=name, eager_start=True)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2081, in _async_forward_entry_setup
    await entry.async_setup(self.hass, integration=integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 134, in async_setup_entry
    return await component.async_setup_entry(entry)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 196, in async_setup_entry
    return await self._platforms[key].async_setup_entry(config_entry)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 333, in async_setup_entry
    return await self._async_setup_platform(async_create_setup_awaitable)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 363, in _async_setup_platform
    awaitable = create_eager_task(awaitable, loop=hass.loop)
  File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task
    return Task(coro, loop=loop, name=name, eager_start=True)
  File "/config/custom_components/solarman/sensor.py", line 89, in async_setup_entry
    _do_setup_platform(hass, entry.options, async_add_entities)
  File "/config/custom_components/solarman/sensor.py", line 54, in _do_setup_platform
    inverter = Inverter(path, inverter_sn, inverter_host, inverter_port, inverter_mb_slaveid, lookup_file)
  File "/config/custom_components/solarman/solarman.py", line 31, in __init__
    with open(self.path + self.lookup_file) as f:
```

gedger, 2024-07-03:
I have the same, I think it's due to the change startup regime that HA now uses.

davidrapan, 2024-07-03:
> I have the same, I think it's due to the changed startup regime that HA now uses.

Yes it is. I have fixed it in my integration overhaul by using aiofiles and its async method.