StephanJoubert/home_assistant_solarman #621: Logger Solarman - randomly disconnected several times a day since 1.5.xx

Aqerkz, 2024-07-07:
Hello team, 

i had to rollback to 1.4.3 to avoid connection issues.
logger serial 23xxxxxxxx (V4 ? )
Sofar Hybrid 5kw

With 1.4.3 everything is working as expected.

Can you have a look on it please ? 
Thank you for your help.

hildevo, 2024-07-13:
It is even worse here. Since last week, Solarman is completely disconnected in Home Assistant. I tried to downgrade to 1.4.3 but that didn't work.

FajnyRadzio, 2024-07-13:
> It is even worse here. Since last week, Solarman is completely disconnected in Home Assistant. I tried to downgrade to 1.4.3 but that didn't work.

Yep, same here. It's not even reacting to IP change.

Ady310, 2024-07-21:
> It is even worse here. Since last week, Solarman is completely disconnected in Home Assistant. I tried to downgrade to 1.4.3 but that didn't work.

same problem to me

hildevo, 2024-07-21:
I deleted the integration and the HACS module, reinstalled everything. Works again now.

PV-Joe, 2024-07-21:
> > It is even worse here. Since last week, Solarman is completely disconnected in Home Assistant. I tried to downgrade to 1.4.3 but that didn't work.
> 
> same problem to me

You have to delete the Solarman folder before downgrading and restart.

Maybe you would like to try this Solarman integration? I think Stephen’s integration is not maintained regularly anymore.

[https://github.com/davidrapan/ha-solarman](url)