StephanJoubert/home_assistant_solarman #633: DEYE status_connection changed to Disconnected

huynhvanphuc2005, 2024-07-29:
Dear Support,

I am using Deye solar power device with logger code "293299XXX" I followed the config instructions "https://github.com/StephanJoubert/home_assistant_solarman/blob/main/README.md" 

The result is that the device says "DEYE status_connection changed to Disconnected" I have tried many ways but it doesn't work. (I have attached images 1-2-3) in the article. 

I hope that if anyone knows about my above case, please guide me.

Sincerely thank you.

[image]
[image]
[image]

SechM, 2024-07-30:
I have this problem too. Sensors don't get parameters from Deye 5 kWt inverter. Maybe solarman have problem with Firmware version LSW3_32U_5406_1.06?!

huynhvanphuc2005, 2024-07-30:
I've already solved that problem. Just open port 8899 directly on the router and the HAS system will recognize the Inverter's logger. 

Good luck.

SechM, 2024-07-30:
I don't understand Why need  open port 8899 on the home router if all equipments located in the local network when ports opened by default.  But  that the Home Assistant does not see the equipment is a fact. Through the official applications of Deye and Solarman, inverters are easy connected. Thank you for your quick answer but with regret I   did not solve the problem yet.(

SechM, 2024-07-30:
Apparently, you have experienced interruptions in your work. I did not start the connection initially. Inverter SUN-5K-SG03LP1-EU, Firmware version LSW3_32U_5406_1.06, logger code "29184XXXX"

SechM, 2024-07-31:
I have found a solution. The corrected code in this repository https://github.com/davidrapan/ha-solarman  . Special thanks to David.