StephanJoubert/home_assistant_solarman #643: HYD 6000-EP sofar + SOLARMAN INTEGRATION + HA = imposible to connect

miralpeix, 2024-08-10:
Dear friends

After a week trying different things to connect HA with my solarman inverter… I need some help. 

Current setup:

Inverter: HYD 6000-EP sofar

[image]

As first option I have tried to use the SOLARMAN INTEGRATION, however all devices are show as “unknown”.

[image]

Following another recomendations I have tried in oder to solve:

1. Different versions of solarman integrations (1.5… 1.4…)
2. Different yalm configurations (sofar_lsw3.yalm, deye_hybrid.yalm….)
3. Due to possible  problems in HA, I have reinstalled in different raspberrys…
4. I have tried alternative versions of integrations of solarman…
5. I have spoken with sofar to update the logger FW to the latest version available

Also I have trie another alternative solutions like MQQT + SUN SINK: https://www.youtube.com/watch?v=BJzJSi6TZMk but didn’t work too.

I feel like it is impossible to connect with my logger from any way using HA.

Some recommendations?? I’m about to surrender after this week…

PV-Joe, 2024-08-10:
You might want to try this fork:
https://github.com/davidrapan/ha-solarman

I think you have to choose:
sofar_hyd3k-6k-es.yamL

Make sure to fully delete any other integration before and delete all sensors belonging to that old integration under:
Developer Tools > Statistics

Good luck!

miralpeix, 2024-08-10:
Thanks a lot for your help! I have reinstall HA and HACS in order to sure that no there're old integrations.
This are some some screenshots. Some suggestion? It is like there's no way to connect.

[image]
[image]

PV-Joe, 2024-08-10:
I well, it was well worth a try.

Open an issue on David’s site. He is usually extremely fast and helpful.

Make sure to attach the log file as he will ask for it anyway.

Go here:
https://github.com/davidrapan/ha-solarman/issues

lepereded, 2024-09-02:
Hello, i ve got a 6000 EP , it works with 
sofar_g3hyd.yaml

regards
[image]

[image]

[image]