StephanJoubert/home_assistant_solarman #650: DEYE SUN-M80G4-EU-Q0 not able to connect! Which Profile?

viertel, 2024-08-27:
Hello all,

i cannot connect to Homeassistant Solarman with the  deye_string.yaml Profile.
In the PC Webinterface (No Homeassistant) I can see the Data logger with all information.
Where the used configuration is stored?
Do i need to modify the configuration?
I added my IP address and serial number in the UI.
[image]

Thanks in advance for help!

PV-Joe, 2024-08-27:
Try this more advanced version:

https://github.com/davidrapan/ha-solarman/

Wellagent68, 2024-08-27:
I have same problem. He suddenly lost contact. The integration is loaded, but all entities are unavailable in Solarman. What happened? How can I fix it? Thanks in advance for your advice.

9Tzw3n6, 2024-08-28:
Hi guys. Had the same issue. 
I had to connect to the inverter AP and to do the "quick set" and "wizard" again.
Dont know why, after that it worked again

Wellagent68, 2024-08-28:
Hello, thanks for your answer. 
Could you please help me how I can connect to inverter (DEYE) AP and run this quick set/wizard?
My son configured it and he is not reachanble now :(
Thanks.
BR, Zsolt

9Tzw3n6, 2024-08-28:
> Hello, thanks for your answer. Could you please help me how I can connect to inverter (DEYE) AP and run this quick set/wizard? My son configured it and he is not reachanble now :( Thanks. BR, Zsolt

Sure, I can try.
Do you know if your son has changed the access point name/password of the inverter?
Is the IP address for the inverter fix?

Since mine is far back in the garden, I'll have to go there.
Search for the Wifi networks.
The WLAN from the inverter is normally called AP_Number. Unless your son has changed it or hidden it :D
The WLAN key or password is on the sticker on your inverter. Unless your son has changed it :D
If everything fits and you are connected to the inverter WLAN - you can call up the IP address of the inverter in the browser. Be aware you dont have Internet Connection when you are connected to the inverter WIFI
I think the default for Deye is http://10.10.100.254/
You will then need your admin password.
Default user is admin and password is also admin, unless it has been changed.

If you are logged in you will see a menu on the left.
Here you can click on Wizard.
Then select your home WIFI network name and enter the WIFI password.
Then enter the IP you have assigned to the inverter. It is important to assign a fixed IP.
In my case, I have assigned 192.168.178.126 for the inverter in the router.
I then enter this in the menu.
Then deactivate IP address auto assignment.
If nothing else is configured in your home, enter the address of your router under Gateway.
Normally this would be 192.168.178.1
You can actually leave out DNS.
Subnet mask 255.255.255.0

I think that's it, you can ignore the options with Hide AP SSID and Co for now and leave them unchecked.

The inverter will then restart once.
Then play the same game again.
Connect to the WIFI AP of the inverter.
Call up the website http://10.blabla/.
Log in
Quickset and again select your home WIFI name, enter the password for the WIFI and set the same IP settings as for Wizard.

I hope I was able to help halfway :'D Otherwise it's best to wait until your son is available

9Tzw3n6, 2024-08-28:
> You are very kind :) Thanks.
> 
> I will try it. One more question. I use Solarman Business mobile app and it works correctly so inverter has WIFI access I think. So I have to do this WIFI part independently from it has internet access?

To be honest, no idea.

I don't have the Solarman App
My WLAN settings are actually correct aswell
I can also access the panel's website.

Only HomeAssistant doesn't want to.
And then when I clicked through the wizard again... it worked :'D