StephanJoubert/home_assistant_solarman #651: Wrong battery current after inverter firmware upgrade

silviudc, 2024-09-03:
My Deye SUN-16K-SG01LP1-EU had it's firmware upgraded today and now the battery current value is incorrect.

Control Board Version No. from 36898 to 36903
Communication Board Version No. from 50017 to 50034

The value is now 10 times less than what the inverter is showing so I think the register changed it's value and scale needs to be adjusted in the deye_hybrid.yaml?

```    - name: "Battery Current"
      class: "current"
      state_class: "measurement"
      uom: "A"
      scale: 0.01
      rule: 2
      registers: [0x00BF]
      icon: 'mdi:battery'
```

PV-Joe, 2024-09-03:
Do you think you can also post this valuable information in the new Solarman Integration here?

https://github.com/davidrapan/ha-solarman

This new advanced version is very well maintained at the moment and has excellent support!

See you there ☀️

silviudc, 2024-09-03:
Changing `scale: 0.1` fixes it so it must have changed in the new firmware

Bruno999B, 2025-04-02:
Hello, do you also have a problem with the DC part temperature display in Solarman? Mine displays -41°C!? The AC and battery temperatures are correct.