StephanJoubert/home_assistant_solarman #654: Add S5-GC30K support

poligor84, 2024-09-09:
Good day.
Please add support for my grid inverter S5-GC30K support.
This is my inverter in Solis cloud:
[image]
and not all sensors are displayed in your application:
[image]
[image]
Thank you.

davidrapan, 2024-09-09:
Hi @poligor84 did you tried load it with every profile there?

poligor84, 2024-09-09:
> Hi @poligor84 did you tried load it with every profile there?

What do you mean by "profile"?

poligor84, 2024-09-09:
> Hi @poligor84 did you tried load it with every profile there?

[image]

davidrapan, 2024-09-09:
yes, I would suggest you to try it with every one of them...

davidrapan, 2024-09-09:
yes the solis, sofar and zcs are in some cases very similar...

poligor84, 2024-09-09:
> yes all the solis, sofar and zcs are in some cases very similar...

Tried. The situation in general has not improved!

davidrapan, 2024-09-09:
Uhh, so you can also try this integration: [davidrapan/ha-solarman](https://github.com/davidrapan/ha-solarman) there is even more of them... otherwise it has to be added as a new one then, you could try to ask your manufacturer to give you modbus rtu documentation for your device...

poligor84, 2024-09-09:
[image]
[image]

davidrapan, 2024-09-09:
The profile is missing PV3 entities. 😉 I can try to add them in my integration if you help me locate those registers. 😉

poligor84, 2024-09-09:
> Uhh, so you can also try this integration: [davidrapan/ha-solarman](https://github.com/davidrapan/ha-solarman) there is even more of them... otherwise it has to be added as a new one then, you could try to ask your manufacturer to give you modbus rtu documentation for your device...

[image]
[image]
[image]
[image]
I can't join for some reason?!

davidrapan, 2024-09-09:
Enable debug and show me the logs please

davidrapan, 2024-09-09:
You have to disable the Solarman protocol passthrough mode! It's only for ethernet sticks!

poligor84, 2024-09-09:
[image]

davidrapan, 2024-09-09:
Yes.

poligor84, 2024-09-09:
to disable

poligor84, 2024-09-09:
> Yes.

no work

davidrapan, 2024-09-09:
Click on enable debug logging, try to add device. Wait few seconds and then click on it again. It will download a file. Post the file here please.

davidrapan, 2024-09-09:
?

poligor84, 2024-09-09:
> ?

Maybe I have a conflict between two versions? and can't connect?

davidrapan, 2024-09-09:
The on you downloaded as last is "active" so no. I told you what to do so we can find out what's wrong. I need the debug logs please. 😉

poligor84, 2024-09-09:
> The one you downloaded as last is "active" so no. I told you what to do so we can find out what's wrong. I need the debug logs please. 😉

[home-assistant_solarman_2024-09-09T09-18-59.702Z.log](https://github.com/user-attachments/files/16928371/home-assistant_solarman_2024-09-09T09-18-59.702Z.log)

davidrapan, 2024-09-09:
Yeah it has problems with some requested registers. You have to try use some different profile.

poligor84, 2024-09-09:
> Yeah it has problems with some requested registers. You have to try use some different profile.

Thanks, working on the profile 3P-4G!

poligor84, 2024-09-09:
[image]

davidrapan, 2024-09-09:
Wait for new version (I already committed the file if you know how to "replace" it manually) where will be added profile with more PV inputs. 😉

poligor84, 2024-09-09:
Ok, thanks.

davidrapan, 2024-09-09:
@poligor84 Beta version [v24.09.09](https://github.com/davidrapan/ha-solarman/releases/tag/v24.09.09) contains profile named: solis_3p-4g+.yaml. You have to allow installing beta versions in HACS under 3 dots menu.

poligor84, 2024-09-10:
Thanks, I will!
[image: image.png]

It works!
Thank you!!!

jedifr, 2025-03-06:
Hello,
I have the same inverter, but my connection key is WIFI.
My concern is that I am not able to insert the serial number of the key because it does not match the expected number type.
My number is type: 5A12338417B0AFF6

Do you have an idea?
THANKS

poligor84, 2025-03-06:
I don't quite understand what you mean?
You can't connect to HA?

jedifr, 2025-03-07:
I can't introduice my serial number key with good numers like 5A12338417B0AFF6

[image]